rtl-airband/RTLSDR-Airband #85: Airspy Audio Problems

vaisala, 2018-02-20:
Using Airspy as SDR receiver for AM with the example file of the wiki (default sample_rate) everything works fine, however if I set the sample_rate 10MBPS (sample_rate = 10000000) the audio is accelerated, similar to a fast forward on a audio player approximately at twice the actual voice speed.
[121675__20180220_11.zip](https://github.com/szpajder/RTLSDR-Airband/files/1741720/121675__20180220_11.zip)

szpajder, 2018-02-20:
https://github.com/szpajder/RTLSDR-Airband/wiki/Troubleshooting#audio-speed-is-wrong

szpajder, 2018-02-20:
https://github.com/szpajder/RTLSDR-Airband/wiki/General-configuration-file-structure#specifying-frequency-values

"10" means 10 Hertz. "10.0" means 10 Megahertz.

vaisala, 2018-02-20:
Thanks for your prompt reply but Setting  sample_rate = 10.0; just get the same wrong speed.

szpajder, 2018-02-20:
Run this for a couple of seconds:
```
SoapySDRUtil --args=driver=airspy --rate=10000000 --direction=RX
```
What sample rate does it show?

vaisala, 2018-02-20:
`f900 RTLSDR-Airband # SoapySDRUtil --args=driver=airspy --rate=10000000 --direction=RX
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

[INFO] Using format CS16.
Stream format: CS16
Num channels: 1
Element size: 4 bytes
Begin RX rate test at 10 Msps
Starting stream loop, press Ctrl+C to exit...                                                    6.55799 Msps     26.232 MBps
6.57978 Msps    26.3191 MBps
6.58269 Msps    26.3308 MBps
6.58473 Msps    26.3389 MBps
6.59 Msps       26.36 MBps
`

szpajder, 2018-02-20:
So you are only getting approx 6.5 Msps, not 10 Msps. This causes the audio speed problem.

Perhaps you have insufficient CPU power or USB bandwidth for that large sampling rate. Are you running this on a Raspberry Pi or similar platform? Can you try this out on a Linux PC?

starvald, 2018-02-20:
you can add bitpack=true to the device string to save 25% usb traffic.
might help.

szpajder, 2018-02-20:
The main problem with Airspies is that these devices do not provide I/Q data directly. It has to be converted into I/Q in software, which hogs the CPU immensely. I'm barely able to run Airspy Mini on a RPi v2 with the lowest sampling rate (3 Msps).

vaisala, 2018-02-20:
My fault trying to run RTL-Airband on a low power pc (again) on a old i3 everything runs fine with correct speed voice.

> root@i3:/# SoapySDRUtil --args=driver=airspy --rate=10000000 --direction=RX
> ######################################################
> ## Soapy SDR -- the SDR abstraction library
> ######################################################
> 
> [INFO] Using format CS16.
> Stream format: CS16
> Num channels: 1
> Element size: 4 bytes
> Begin RX rate test at 10 Msps
> Starting stream loop, press Ctrl+C to exit...                                                              9.98028 Msps     39.9211 MBps
> 9.9902 Msps     39.9608 MBps
> 9.9935 Msps     39.974 MBps
> 9.99514 Msps    39.9806 MBps
> 9.99613 Msps    39.9845 MBps
> 9.9968 Msps     39.9872 MBps

closed 2018-02-20 by vaisala