rtl-airband/RTLSDR-Airband #87: Errors while using rtl_airband

norbeta, 2018-03-06:
Hey, 

This is a really nice software that I've used for a while now. 

I sometimes get these errors but the program continues to run: 
`rtlsdr_demod_write_reg failed with -9`

But when I get this error...
`r82xx_write: i2c wr failed=-9 reg=17 len=1
r82xx_set_freq: failed=-9`
... the program hangs (after version 2.3.0). It will not exit with ctrl+c, I must use kill -9 to quit the program completely. 

I've tried with three different rtl-sdr.org-dongles, but all of them seem to produce the same error.

szpajder, 2018-03-06:
These error messages are not produced by rtl_airband but rather by librtlsdr library, which has encountered a problem communicating with the device.

You have found that the issue is not related to a particular dongle, so maybe it's related to a particular computer? Have you tried reproducing it on another machine?

Is it a Pi or a PC? What Linux distribution and version?

norbeta, 2018-03-07:
I have not tried another computer yet. 

It is a PC, running Debian 9. I may try to set it up on another computer, but I think its still a little odd that the program didn't crash on this error until version 2.3.0.

szpajder, 2018-06-09:
Closing this, as it's not a thing which can be fixed in rtl_airband.

closed 2018-06-09 by szpajder