rtl-airband/RTLSDR-Airband #88: Unnecessary squelch deactivation in scan mode

j4cekc1dz, 2018-03-20:
Hi,

I noticed, that squelch is unnecessary deactivated in nfm scan mode for one demodulation iteration and causes unpleasant crackle per 1 second (few seconds). Seems like fparms->agcavgslow in the first demodulation iteration after frequency has been changed is related to the previous frequency. 

Waterfall:
  24/ 23  149.287
  52/ 23  149.287
  50/ 49  149.900
  81/ 49  149.900
  71/ 22  148.800
  25/ 22  148.800
  23/ 23  149.112
  73/ 23  149.112
  73/ 25  149.137
  27/ 25  149.137
  26/ 23  149.188
  24/ 23  149.188

Program works on RPi3 and has been built with NFM=1 from the latest commit in the master branch.

szpajder, 2018-03-20:
I often do scanning with NFM and I don't have this problem.

What type of receiver is this? RTLSDR? Did you capture this waterfall when the squelch was closed? If yes, then your gain might be too high (which could cause spurious signals and be the cause of the issue you've described). When the squelch is closed, the signal level should be more or less equal to the noise floor level.

j4cekc1dz, 2018-03-21:
First of all sorry, because it was output, where signal/noise factor was changed from 3.0f to 3.6f. With this change issue occurs very rare. As second workaround, frequencies has been ordered from the biggest to the least background noise.

Anyway, seems that in my area at +- 149.9MHz some bigger noise exists than in lower frequencies.
I'm using R280T + RTL2832U

After factor has been restored to 3.0, I managed to reproduce issue:

 148.800                                                                       

  23/ 22  149.137
  24/ 21  149.188
  23/ 21  149.188
  21/200  149.900
  59/200  149.900
  54/ 20* 148.800
  22/ 20  148.800
  21/ 20  149.112
  47/ 20  149.112
  46/ 29  149.287
  37/ 29  149.287

szpajder, 2018-03-21:
Is the high background noise real or it's just a result of receiver saturation or an intermod or aliasing product? Aliasing often happens when running the RTL on the default sampling rate of 2.56 Msps. Check out if you are able to get a cleaner spectrum with a sample rate of 2.4 Msps. See here for more info: https://github.com/szpajder/RTLSDR-Airband/wiki/Troubleshooting#aliasing-artifacts-at-high-sampling-rates . If this is not the case, then instead of fiddling with auto squelch, you can set the squelch manually for each scanned frequency - refer to https://github.com/szpajder/RTLSDR-Airband/wiki/Manual-squelch-setting for an example.

General note - this program is not really meant for scanning. The demodulator and scanning controller run completely independently - neither of them knows what the other one is doing at the moment. The demodulator does not know precisely, which scanned channel the given sample batch came from. This is an inherent limitation of batching FFTs into large blocks to be computed on the GPU.

Given that your channels are so close to each other in frequency, why don't you just use multichannel mode? If you want them all in a single audio stream, just use the mixer feature.

j4cekc1dz, 2018-03-21:
I'm using 1.6Msps and gain 20. 
Regarding general note - I know this after code analyzing. Demodulator and controller_thread work independently but I didn't know, it's a bug or program/hardware limitation. Anyway, frequency ordering and factor increasing work for me. Maybe I will try to synchronize it later but not sure is there any sense.

I tried multichannel mode. Unfortunately distance from transmitter was decreased.
Settings I used (the same channels):
min_freq = 148.8MHz
max_freq = 149.9MHz
fft_size = 512
sample_rate = 1.6Msps
centerfreq = 149MHz + sample_rate/fft_size

Maybe should be
centerfreq = 149MHz + (sample_rate/fft_size/2) ?

szpajder, 2018-03-21:
Too low - 149.9 MHz won't fit in the dongle bandwidth. Centerfreq, as the name says, shall be more or less in the center of your band of interest, ie. somewhere around 149.35 MHz.

j4cekc1dz, 2018-03-28:
Multichannel mode works very well. I wrote modulation = "nfm" in device instead of channel part in config file and it was the issue regarding multichannel. 

Thank you and please close this issue.

szpajder, 2018-03-28:
Thanks for the feedback. Closing.

closed 2018-03-28 by szpajder