rtl-airband/RTLSDR-Airband #90: Automatic squelch

PeterAxelsson, 2018-06-08:
Add a function to adjust/set the automatic noise level factor.

Best regards
Peter Axelsson

PeterAxelsson, 2018-06-09:
Is it possible to change the level direct in the code?

Best regards
Peter Axelsson

szpajder, 2018-06-09:
Yes it is possible. Find this line in rtl_airband.cpp (it's somewhere around line 470):

    float sqlevel = fparms->sqlevel >= 0 ? (float)fparms->sqlevel : 3.0f * fparms->agcmin;

Change the 3.0f value to whatever you wish.

PeterAxelsson, 2018-06-10:
Brilliant!!!
4.0f works better for me.

Best regards
Peter Axelsson

szpajder, 2018-08-18:
From my experience, 4.0 is not sensitive enough (cuts off too many weak and distant transmissions), so I'd rather keep this at 3.0. And I'm not convinced if there's really a need for a config knob for this - auto means auto and that's it. In cases when auto does not work correctly, you may use manual mode.

closed 2018-10-14 by szpajder

jlpapple, 2018-10-24:
I need a more sensitive auto squelch for my airspyHF+, can I adjust this lower to 2.5f or 2.0f?  Do I need to do this before compiling?

I can’t effectively use the manual mode because whole numbers, even “1”, is too large a threshold when my signal is only 2-4.  Thanks.

szpajder, 2018-10-24:
Feel free to experiment. However it would be better to investigate if the
signal is indeed so low or if there is still something something wrong with
processing samples from HF+ by rtl_airband or SoapySDR driver. You
mentioned that you've got it to work with GQRX - so what's the typical
noise floor level and signal level in your channels of interest (in dBFS),
as shown by GQRX?

jlpapple, 2018-10-24:
Just recompiled at 2.0f for squelch, will test and monitor.  

Typical noise floor is a constant 103-108dBFS.  The signal on the transmissions in question are between 80-90dBFS.  

I'll need to monitor more to get more precise signal values.