rtl-airband/RTLSDR-Airband #91: Compile error on Raspberry Pi 1 model B

OZ1SEJ, 2018-06-13:
When I issue the command

    make PLATFORM=rpiv1

I get this reponse

```
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.0.1-dirty\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -march=armv6zk -mfpu=vfp -ffast-math  -DWITH_RTLSDR   -c -o rtl_airband.o rtl_airband.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.0.1-dirty\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -march=armv6zk -mfpu=vfp -ffast-math  -DWITH_RTLSDR   -c -o input-common.o input-common.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.0.1-dirty\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -march=armv6zk -mfpu=vfp -ffast-math  -DWITH_RTLSDR   -c -o input-helpers.o input-helpers.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.0.1-dirty\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -march=armv6zk -mfpu=vfp -ffast-math  -DWITH_RTLSDR   -c -o output.o output.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.0.1-dirty\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -march=armv6zk -mfpu=vfp -ffast-math  -DWITH_RTLSDR   -c -o config.o config.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.0.1-dirty\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -march=armv6zk -mfpu=vfp -ffast-math  -DWITH_RTLSDR   -c -o util.o util.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.0.1-dirty\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -march=armv6zk -mfpu=vfp -ffast-math  -DWITH_RTLSDR   -c -o mixer.o mixer.cpp
make -C hello_fft
make[1]: Entering directory '/home/pi/RTLSDR-Airband/hello_fft'
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.0.1-dirty\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -march=armv6zk -mfpu=vfp -ffast-math  -DWITH_RTLSDR -I..   -c -o mailbox.o mailbox.c
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.0.1-dirty\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -march=armv6zk -mfpu=vfp -ffast-math  -DWITH_RTLSDR -I..   -c -o gpu_fft.o gpu_fft.c
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.0.1-dirty\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -march=armv6zk -mfpu=vfp -ffast-math  -DWITH_RTLSDR -I..   -c -o gpu_fft_twiddles.o gpu_fft_twiddles.c
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.0.1-dirty\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -march=armv6zk -mfpu=vfp -ffast-math  -DWITH_RTLSDR -I..   -c -o gpu_fft_shaders.o gpu_fft_shaders.c
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.0.1-dirty\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -march=armv6zk -mfpu=vfp -ffast-math  -DWITH_RTLSDR -I..   -c -o gpu_fft_base.o gpu_fft_base.c
gpu_fft_base.c: In function ‘unsigned int gpu_fft_base_exec_direct(GPU_FFT_BASE*, int)’:
gpu_fft_base.c:65:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (q=0; q<num_qpus; q++) { // Launch shader(s)
               ~^~~~~~~~~
gpu_fft_base.c:72:50: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (((base->peri[V3D_SRQCS]>>16) & 0xff) == num_qpus) break; // All done?
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
gpu_fft_base.c:54:17: warning: unused variable ‘t’ [-Wunused-variable]
     unsigned q, t;
                 ^
ar r hello_fft.a mailbox.o gpu_fft.o gpu_fft_twiddles.o gpu_fft_shaders.o gpu_fft_base.o
ar: creating hello_fft.a
make[1]: Leaving directory '/home/pi/RTLSDR-Airband/hello_fft'
as   -o rtl_airband_vfp.o rtl_airband_vfp.s
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.0.1-dirty\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -march=armv6zk -mfpu=vfp -ffast-math  -DWITH_RTLSDR   -c -o input-rtlsdr.o input-rtlsdr.cpp
input-rtlsdr.cpp:30:21: fatal error: rtl-sdr.h: No such file or directory
 #include <rtl-sdr.h>
                     ^
compilation terminated.
<builtin>: recipe for target 'input-rtlsdr.o' failed
make: *** [input-rtlsdr.o] Error 1

```

szpajder, 2018-06-13:
Just read the error message:

    fatal error: rtl-sdr.h: No such file or directory

You don't have librtlsdr installed properly.

OZ1SEJ, 2018-06-13:
Doh! I am now officially unable to follow even very well-written instructions. Mea culpa. Sorry!

closed 2018-06-13 by OZ1SEJ