rtl-airband/RTLSDR-Airband #92: Improve selectivity and reduce intermodulation distortion

saedelman, 2018-06-29:
We have an AWOS on the field operating at 122.55MHz and a Unicom ATF frequency at 123.50MHz (almost 1MHz frequency separation) and we routinely hear traffic in the circuit transmitting on 123.50MHz mixed in with the AWOS feed. 

A regular superheterodyne radio has no problem with selectivity, but it appears not the case with RTL-Airband. How can we improve selectivity and rejection of out of band transmissions? We're using a premium SDR with E4000 tuner and have the gain set to 20. I suspect the limited dynamic range of the tuner is the issue and the relatively strong signals are causing clipping and splatter across the capture bandwidth. 

I understand that I can reduce this by lowering the gain, but is it possible to tighten the FFT window (at the expense of higher CPU usage)? Where in the code can this be adjusted?

szpajder, 2018-06-29:
> A regular superheterodyne radio has no problem with selectivity, but it
> appears not the case with RTL-Airband.
>
It's because a regular superheterodyne radio is a narrowband device by
design, while your premium SDR with E4000 tuner is a wideband device.

How can we improve selectivity and rejection of out of band transmissions?
>
By reading the manual and following instructions provided there:

https://github.com/szpajder/RTLSDR-Airband/wiki/Troubleshooting#audio-bleeding-between-channels

> We're using a premium SDR with E4000 tuner and have the gain set to 20. I
> suspect the limited dynamic range of the tuner is the issue and the
> relatively strong signals are causing clipping and splatter across the
> capture bandwidth.
>
Most probably. You may verify this with any SDR console (GQRX, SDRSharp,
whatever). If you see junk on the spectrum, RTLSDR-Airband will demodulate
and play this junk through your speakers. No magic here.

> I understand that I can reduce this by lowering the gain, but is it
> possible to tighten the FFT window (at the expense of higher CPU usage)?
> Where in the code can this be adjusted?
>
Again, please read the manual - it has been written for you:

https://github.com/szpajder/RTLSDR-Airband/wiki/Tweaking-sampling-rate-and-FFT-size

saedelman, 2018-06-29:
Wow. Not sure how I missed this. Thank you.

closed 2018-08-18 by szpajder