rtl-airband/RTLSDR-Airband #94: RTLSDR device #0: async read failed, disabling

harf18, 2018-07-10:
Hi,

I have an issue with RTLSDR-Airband

## Description

it works fine during 20 to 40 secondes (ie. I heard sound on icecast's stream) then it crash.

In foreground execution, the message is *cb transfer status: 1, canceling...*
In background execution, the message is *RTLSDR device #0: async read failed, disabling*

is my dongle crap ?

## My configuration

- I use airband with a raspberry pi 3. 
- The dongle is a RTL2838
- I compile airband with **rpiv2** option
- My output is configured to send stream to icecast
- I use a usb power adapter

```
devices:  
({  
  type = "rtlsdr";  
  index = 0;  
  gain = 25;  
  centerfreq = 121.77;  
  correction = 80;  
  channels:  
  (  
    {  
      freq = 121.77;  
      outputs: (  
        {  
          type = "icecast";  
          server = "icecast.myrealdomain.com";  
          port = 8000;
          mountpoint = "atr";
          name = "Tower";
          genre = "ATC";
          username = "source";
          password = "my-real-password";
        }
      );
    }
  );
}
);
```

szpajder, 2018-07-10:
Your config is fine and your dongle is probably OK too. Go google "cb transfer status: 1, canceling" and you'll get tons of results. It's a common issue which is most often related to underpowered USB device. rtl_airband cannot do anything about it, but you can:

- check the voltage between 5VDC and GND pins in USB port of the Pi (or easier - between pins 4 and 6 on the GPIO head). It shall be close to 5V. If it's 4.8 or lower - expect problems.

- if the voltage is too low, replace the USB power cable (the thicker wires, the better)

- eliminate unnecessary USB extension cables

- get the power supply which can deliver 5V (or a even a bit more) when loaded.

- reconnect the dongle to a powered USB hub (above remarks above USB cables still apply).

harf18, 2018-07-10:
Thank you for this detailed answer.
I'll go get search a better power supply.

harf18, 2018-07-12:
It's ok with a better power supply. Thanks a lot !

closed 2018-07-12 by harf18