rtl-airband/RTLSDR-Airband #96: Build fails on Rpi3 - Raspbian Stretch 4.14

Telefonorosso, 2018-08-01:
Dear sir,
Cloning into latest git and issuing a "PLATFORM=rpiv2 make" got me this output:

**_g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.0.1\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv7-a -mfpu=neon-vfpv4 -mfloat-abi=hard -ffast-math  -DWITH_RTLSDR   -c -o rtl_airband.o rtl_airband.cpp
rtl_airband.cpp:61:25: fatal error: libconfig.h++: No such file or directory
 #include <libconfig.h++>
                         ^
compilation terminated.
<builtin>: recipe for target 'rtl_airband.o' failed
make: *** [rtl_airband.o] Error 1_**

Any help will be highly appreciated!
Regards,
FT

szpajder, 2018-08-01:
You don't have libconfig installed.

Got through this:
https://github.com/szpajder/RTLSDR-Airband/wiki/Dependencies ?

Telefonorosso, 2018-08-01:
Thank you for the realtime reply!
Now I get this:

**_input-rtlsdr.cpp:30:21: fatal error: rtl-sdr.h: No such file or directory
 #include <rtl-sdr.h>_**

Telefonorosso, 2018-08-01:
Solved with

**_pt-get install librtlsdr-dev_**

my bad, didn't read the page you suggested.
I was in a rush to use the software!!!

closed 2018-08-01 by Telefonorosso