rtl-airband/RTLSDR-Airband #100: App hangs with message "Failed to set centerfreq for RTLSDR device"

risticsasa011, 2018-09-20:
Hello, 

after starting, RTLSDR-Airband runs for a 1-2 hours  (I keep it in a screen), but then hangs with this message: 
"Failed to set centerfreq for RTLSDR device". I need to kill -9 to be able to restart...

Any idea why?

szpajder, 2018-09-20:
Not too many details...

Is this on a RPi? If yes, then please rule out power problems first (
https://github.com/szpajder/RTLSDR-Airband/issues/94).

risticsasa011, 2018-09-20:
Sorry, no, it's on a PC with regular power supply.
I'm waiting for it to crash again to send more details...

risticsasa011, 2018-09-20:
Hello, more details folow:
---
r82xx_write: i2c wr failed=-9 reg=10 len=1
r82xx_set_freq: failed=-9
Failed to set centerfreq for RTLSDR device #0: error -9
Closing connection to $streaming_server_ip
All receivers failed, exiting
Cleaning up
---
CRTL+C is unable to exit process. I need to do "killall -9 rtl_airband" to be to restart it.
I don't see anything in system logs.

szpajder, 2018-09-20:
Which libusb version do you have? And which librtlsdr version/fork?

szpajder, 2018-09-20:
1.0.17? This is very old. I guess it's Ubuntu Trusty?

I would recommend an upgrade. libusb 1.0.17 and earlier have many bugs (eg.
https://github.com/szpajder/RTLSDR-Airband/wiki/Troubleshooting#rtlsdr_demod_write_reg-failed-with--1
).

The problem you've described is not a bug in rtl_airband; it just wants to
change the dongle center frequency but the USB control transfer fails (but
most probably due to libusb bug rather than a hardware problem).

risticsasa011, 2018-09-26:
Meanwhile, since I'm unable to upgrade this Ubuntu for now, I transferred to Raspberry Pi. With libusb 1.0.21 on RTLSDR-Airband is streaming for a few days without any issues.

Thank you for your help and time.

closed 2018-10-14 by szpajder

hypeARM, 2023-05-23:
Hi Charlie, what can be happening?

> [rtlsdr-airband] 2023/05/23 16:08:00 rtlsdr_demod_write_reg failed with -9
[rtlsdr-airband] 2023/05/23 16:08:00 r82xx_write: i2c wr failed=-9 reg=17 len=1
[rtlsdr-airband] 2023/05/23 16:08:00 r82xx_set_freq: failed=-9
[rtlsdr-airband] 2023/05/23 16:08:00 Failed to set centerfreq for RTLSDR device #0: error -9
[rtlsdr-airband] 2023/05/23 16:08:00 Disabling device outputs
[rtlsdr-airband] 2023/05/23 16:08:00 Closing connection to 127,0.0.1:8000/audio
[rtlsdr-airband] 2023/05/23 16:08:00 All receivers failed, exiting
[rtlsdr-airband] 2023/05/23 16:08:00 Cleaning up

This error occurs only in scan mode, with only 2 frequencies being scanned