rtl-airband/RTLSDR-Airband #102: SoapySDR for Airspy HF = "No suitable sample format found"

jlpapple, 2018-10-13:
This is the error message at launch:
**"SoapySDR configuration error: device 'driver=airspyhf,device_id=0': no suitable sample format found"**

My theory is that there is an issue with the SoapySDR Airspy HF+ module sampling format code, see
https://github.com/pothosware/SoapyAirspyHF/issues/1.  Does the Soapy module not explicitly support CU8, CS8, or CS16 in the module - as required by RTL Airband?

**SoapySDRUtil --probe readout:**
  driver=AirspyHF
  hardware=AirspyHF
  device_id=0
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CF32
  Native format: CF32 [full-scale=3.4e+38]
  Antennas: RX
  Full gain range: [0, 0] dB
  Full freq range: [0.009, 31], [60, 260] MHz
    RF freq range: [0.009, 31], [60, 260] MHz
  Sample rates: 0.768 MSps

**SoapySDRUtil --info readout:**
Lib Version: v0.7.0-g5f5a4d8c
API Version: v0.7.0
ABI Version: v0.7
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.7
Module found: /usr/local/lib/SoapySDR/modules0.7/libairspyhfSupport.so (0.1.0-f373896)
Available factories... airspyhf
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

**Relevant RTL Airband config:**
devices:
({
  type = "soapysdr";
  device_string = "driver=airspyhf,device_id=0";
  sample_rate = 768000;
  gain =25.0;
 centerfreq = 160.8;
 correction = 0;

I've been using an RTLSDRv3 dongle successfully for the last two months, but can't get my new Airspy HF+ to work properly via SoapySDR on Raspbian Stretch.  The Airspy is latest firmware (1.7.2), airspy_lib 1.0.9, SoapySDR 0.7.  The AirspyHF is properly recognized with the correct device ID and works fine in GQRX, SDR# via SpyServer.  I'm using V3.0.1 of RTL Airband with rpiv2 ,NFM, Pulseaudio and SoapySDR support selected at compilation.

szpajder, 2018-10-13:
>
> My theory is that there is an issue with the SoapySDR Airspy HF+ module
> sampling format code, see
> pothosware/SoapyAirspyHF#1
> <https://github.com/pothosware/SoapyAirspyHF/issues/1>. Does the Soapy
> module not explicitly support CU8, CS8, or CS16 in the module - as required
> by RTL Airband?
>
Your theory is perfectly correct.

szpajder, 2018-10-14:
CF32 is now supported in unstable branch (https://github.com/szpajder/RTLSDR-Airband/tree/unstable). No config changes needed - just pull, recompile and run, it should pick CF32 automatically if nothing else is supported by the device.

jlpapple, 2018-10-15:
Awesome.  I was successfully able to launch the unstable version, it runs with the CF32 addition.  Thank you!

Unfortunately both the noise and signal waterfalls read a constant "0".  I've checked all antenna connections - and signal is received via GQRX.  

The Airspy HF+ has two antenna ports on them (VHF, HF), and I've tried using the antenna variable - but with no change.  What is the proper antenna config syntax? 0, 1?

I'm suspecting that the issue is the AGC, as the Airspy HF+ does not feature a user-configurable gain, it is all AGC.

What do you think the issue is - and what are the options?  I'd really like to use this $200 device with RTL Airband.

jlpapple, 2018-10-16:
Changing the fullscale value to 1.0f in the Airspy HF+ driver code worked - thank you.

Waterfall values in both columns are a constant "0", with the squelch opening at values of "1" and above for the signal column.  I've never seen a value higher than "6" in the signal column, the most common figures are 1-3.  Noise column with no squelch opening is a constant "0".  Gain adjustments have no effect, sampling rate is default at 0.768, fft is default, no manual squelch in config.

I need to listen for a few more hours, but I do get the sense there is some signal/audio clipping and/or abrupt squelch closing compared to listening with the previous RTLSDR SDR.  Some conversations appear to get cut off - even in strong signal conditions.  Will report back after more monitoring.

Is there any potential benefit to experimenting with the 1.0f fullScale value in the driver?

szpajder, 2018-10-16:
>
> Waterfall values in both columns are a constant "0", with the squelch
> opening at values of "1" and above for the signal column. I've never seen a
> value higher than "6" in the signal column, the most common figures are
> 1-3. Noise column with no squelch opening is a constant "0".
>

Which means the signal level (expressed in sample values, not in absolute
input power) is very low. The full range of sample values are -1 to 1, but
you are probably utilizing a very small fraction of this range.

> Gain adjustments have no effect
>
Because this device does not have configurable gain.

> sampling rate is default at 0.768, fft is default, no manual squelch in
> config.
>

You may try to increase FFT size (like twice or so). You will get better
selectivity and FFT bin values will become a bit higher.

> I need to listen for a few more hours, but I do get the sense there is
> some signal/audio clipping and/or abrupt squelch closing compared to
> listening with the previous RTLSDR SDR. Some conversations appear to get
> cut off - even in strong signal conditions. Will report back after more
> monitoring.
>

Same as above - if sample values are very small, the program is working on
very small float values. If they are close to arithmetic rounding errors,
such problems may arise.

> Is there any potential benefit to experimenting with the 1.0f fullScale
> value in the driver?
>

You can reduce it to, say, 0.5 or even 0.2 to get an artificial "software
gain". But this is cheating - if the device is really able to utilize full
<-1; 1> range, then you may come into a situation when the signal exceeds
the fullScale value and you'll then get clipping and distortion.

> FYI - CPU usage is now 2-3x higher,
>

This is most probably spent in libairsphyhf or SoapySDR library.

I've skimmed through libairspyhf code. It looks like the device produces
signed 16-bit integers which are then converted to <-1; 1> floats in
software, by convert_samples() routine. It also calculates sine and cosine
from libm for each sample. This is computationally expensive.

jlpapple, 2018-10-18:
I increased the fft from 512 to 1024, but the audio begins to sound a bit overdriven and distorted.  It sounds similar to when volume/gain is set too high on an output source.  Increasing the fft to 2048 slows the scan speed so much that it makes it unusable on a Raspberry Pi 3.

I have noticed occasional signal levels between 150-450 when a local pager system transmitter goes off near my home.  The remainder of the time the signal level fluctuates between 0-6 on most transmissions.

I've had to set a manual squelch of 1 for all channels, as some distance stations flutter back and forth with open/closed squelch when not set at 1.  I haven't bothered with modifying the fullScale value to less than 1.0f, will experiment in the near future.

Is there a link that you can send me that explains the function of the -1 to 1 sample values in laymans terms?  Thanks.

szpajder, 2018-10-18:
> I increased the fft from 512 to 1024, but the audio begins to sound a bit
> overdriven and distorted. It sounds similar to when volume/gain is set too
> high on an output source.
>

Given the fact that you have only 768 kbps of bandwidth, FFT=1024 might
indeed give you a bit too narrow bins (channels). So I expected that and
you've verified this in practive.

> I have noticed occasional signal levels between 150-450 when a local pager
> system transmitter goes off near my home. The remainder of the time the
> signal level fluctuates between 0-6 on most transmissions.
>
I've had to set a manual squelch of 1 for all channels, as some distance
> stations flutter back and forth with open/closed squelch when not set at 1.
> I haven't bothered with modifying the fullScale value to less than 1.0f,
> will experiment in the near future.
>

So what's the result actually? Is the audio quality fine now?

Your receiver has a high dynamic range - it can receive very strong signals
without clipping and very weak signals without burying them in the noise.
You also have a very low noise level (near-0 values in the waterfall) which
makes auto squelch in rtl_airband to behave erroneously. The way it works
is that It opens when the signal level exceeds noise level by a factor of
3. So if the noise floor is, let's say, 10 (a typical value on RTL
dongles), it means the signal level must exceed 30 to open the squelch and
then fall down below 30 to close it. But if the noise floor is close to 0,
like in your case, let's say, 0.1, then the signal level threshold for the
squelch is only 0.3. So even a small variance of the noise level may cause
the squelch to open and even a brief decrease of the signal level may cause
it to close. You can simulate the same behaviour with an RTL dongle by
setting its gain close to 0 - and then you can rectify the problem by
increasing the gain. With HF+ you can't do that, because it does not have a
configurable gain. So you should use manual squelch setting instead and you
should be good. No need to cheat by tweaking fullScale in SoapySDR code.

> Is there a link that you can send me that explains the function of the -1
> to 1 sample values in laymans terms? Thanks.
>
An analog-to-digital converter is a device which converts analog voltage
waveform into a series of numbers. It has a limit on the input voltage
range - let's say it's +-100 mV. It means that the ADC can digitize the
input waveform without clipping, if it does not fall below -100 mV or +100
mV at any time. This voltage range is then converted into a number range.
Depending on the SDR type, it can be an integer or a floating point value
and the number range may differ. For example:

- RTL dongle produces unsigned 8-byte values, the range is 0 to 255
(meaning 0 is negative maximum voltage, 255 is positive maximum, 128 is
approx. 0 volts)
- Airspy HF+ produces 16-bit signed values (from -32768 to 32767). This
range is then converted by libairspyhf library to a range of floats between
-1.0 to 1.0.

Have you ever played with recording audio and editing wave files with a
program like Audacity or Sound Forge or CoolEdit? SDRs are quite similar to
sound cards - just sampling rates and frequency ranges are higher.

szpajder, 2018-10-18:
I've filed a bug report for the wrong fullScale value - https://github.com/pothosware/SoapyAirspyHF/issues/3

szpajder, 2018-10-25:
>
> I've ruled out any power supply issues on the Pi, as I've got two official
> PSUs that have tested OK, no undervoltage log messages
>
No log messages are produced on undervoltage condition on the Pi (unless
you've set up your own monitoring and logging facility).

"vcgencmd get_throttled" is the correct way to check for present and past
undervoltage and overtemperature conditions. Details here:
https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=147781&p=972822#p972790

> and my external wattmeter shows the Pi is drawing well under maximum
> specification.
>

I doubt if it can show transient current spikes. This is when USB device
disconnects typically occur.

Of course it might also be a software issue (libairspyhf, Soapy-AirspyHF,
etc). Hard to tell remotely.

> I'll check for USB driver unloading issues and will do a clean install of
> Raspbian to reduce the variables as well. I've also uninstalled Pulse and
> built Airband without Pulse to test.
>

I think this is all irrelevant.

> Still working on getting Raw I/Q's
>
In the other post you've wrote:

103-108dBFS. The signal on the transmissions in question are between
> 80-90dBFS.
>

which agrees with what you see in rtl_airband waterfall.

-100 dBFS is 0.00001 on an absolute sample value scale (on a range between
-1.0 to +1.0), while -80 dBFS is 0.0001. So you are utilizing only a very
very very small portion of the ADC range. Sure, noise level of near-zero
looks spectacular, but what's the benefit of this when the signal value is
also close to 0?

Having an analog gain knob would help here - lower quantization noise,
better numerical properties of all the calculations done at the DSP stage.
Do you have any external amplifiers at hand? Would be nice to try one out.

What about RTL dongles? Do you get any better results on them with
rtl_airband and your signals of interest?

szpajder, 2018-12-09:
So it was not a crash, but an exit on a fatal error. SoapySDR library returned a timeout, because it could not get more samples from the device in a timely manner. As a result, rtl_airband declared the device as unusable and disabled it. Because it had no other operational devices, it exited.

jlpapple, 2018-12-09:
Thanks for the clarification and quick reply. So should I contact the SoapySDR maintainer to address this exit on fatal error?

Based on your experience with SoapySDR, does it seem plausible that this exit on fatal error would only occur on scan mode (and not multichannel mode) when using RTL Airband?

Thank you for any assistance you can provide.

jlpapple, 2018-12-16:
Thanks for the suggestion to independently test with an RTL dongle to narrow down the variables.  I have a RTLSDR and will test it in a few weeks.

In the interim I've had success by adding `Restart=always
RestartSec=3` to the rtl_airband.service configuration file.

It reliably restarts the service from the fatal error in 3 seconds.

szpajder, 2019-02-24:
Closing this, as the "No suitable sample format found" issue is now fixed and https://github.com/pothosware/SoapyAirspyHF/issues/3 is fixed as well.

closed 2019-02-24 by szpajder