rtl-airband/RTLSDR-Airband #103: Scan mode

junglejet, 2018-10-23:
I have installed 3.0.1 and multi mode works great with an RTL dongle. However I cannot get scan mode to work. When I enter "freqs =" in the conf file I get this error message:
`Configuration error: mandatory parameter missing: devices.[0].channels.[0].freq`
When I change "freqs" to "freq" it is not complaining, but it does not seem to scan at all.
Btw, when "centerfreq is not there in scan mode, it is complaining, too:
`Configuration error: mandatory parameter missing: devices.[0].centerfreq`

Here is the complete section that I am trying to get to work
```
devices:
({
  type = "rtlsdr";
  index = 0;
  gain = 30;
  centerfreq = 126.0;
  correction = 0;
  channels:
  (
    {
      freqs = ( 126.850, 125.850, 136.675, 134.250 );
      outputs: (
        {
	  type = "icecast";
	  server = "127.0.0.1";
          port = 8000;
          mountpoint = "atc.mp3";
          name = "ATC";
          genre = "ATC";
          username = "source";
          password = "xxxx";
	}
      );
    }
  );
 }
);
```

szpajder, 2018-10-23:
You need to set `mode = "scan";` in the device section. Refer to https://github.com/szpajder/RTLSDR-Airband/blob/master/config/basic_scanning.conf for an example.

junglejet, 2018-10-23:
Thanks for quick reply, I did not see it in the wiki. The application is super.

closed 2018-12-17 by szpajder