rtl-airband/RTLSDR-Airband #104: Missing scan freq tag when using mixer

ernieek, 2018-11-22:
Hello, 

When i use the mixer to little up amp the audio i lost the function to send Fre-tags to icecast server.

is this a bug? or can this be fixed?

Here my config:

mixers: {
  mixer1: {
   outputs: (
        {

          type = "icecast";
          server = "XXXXXX";
          port = XXXX;
          mountpoint = "StreamX";
          name = "X Arrival - Approach";
          genre = "ATC";
          description = "My local airport - ground feed";
          username = "XXXX";
          password = "XXXXXX";
          send_scan_freq_tags = true;
        }
      );
   };
}

devices:
({
  type = "rtlsdr";
  index = 0;
  #serial = "00000001";
  gain = 2;
  correction = 0;
  mode = "scan";
  channels:
  (
    {
      modulation ="am";
      squelch = 410;
      #afc = 1;
      freqs = ( 122.100, 121.930, 123.180, 124.530, 131.005 );
      labels = ( "Test 1 122.100", "test 2", "Test 3","test 4","test 5 131.005");
      #squelch = ( -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1 );
      outputs: (
        {
          type = "mixer";
          name = "mixer1";
          ampfactor = 3.0;
          send_scan_freq_tags = true;
        }
      );
    }
  );
 }
);

szpajder, 2018-11-22:
>
> When i use the mixer to little up amp the audio i lost the function to
> send Fre-tags to icecast server.
>
Freq tags are not supported on mixer outputs.

 is this a bug? or can this be fixed?
>

This is intended behaviour.
The purpose of mixers is to mix multiple audio streams. Mixing freq tag
streams does not make sense - how would you know which stream do the tags
apply to?

ernieek, 2018-11-22:
hello,

true if you use a mixer for more channels. But i used it for only one scan audio stream.

Now i'm thinking i can use the ampfactor for also output to icecast stream :)

Stuppid i didn't thinking before.

but many thanks :)

Keep up the good work

szpajder, 2018-11-22:
Mixers only mix audio, not tags. Propagating tag streams to mixers is not
implemented and adding this feature does not make a lot of sense because it
would only be useful in this single rare case when the mixer has only one
input stream.

A better option is to have an ampfactor property for each stream, so that
there is no need to use a mixer only to adjust the volume level. This
feature is on TODO list.

ernieek, 2018-11-22:
great to hear :)

is it also on the list, when scanning the tag wil change to scanning in place keeping the old frequncy?

szpajder, 2018-11-22:
Could you please increase the clarity of your question? :)

ernieek, 2018-11-22:
Hello,

What i mean is.

I run the program in dcan mode. So when it stops because there is audio for example the tag is updated to the label or frequency. When it start scanning again the tag is stil showing the old label. It would be nice if the tag is updating to for example "scanning..."

rec9140, 2018-11-22:
>Mixers only mix audio, not tags. Propagating tag streams to mixers is not implemented and adding >this feature does not make a lot of sense because it would only be useful in this single rare case >when the mixer has only one input stream.

Here is what I planned to do:

HEMS uses 

462.950, 155.340, 123.025, 155.220 

So I have a MIXER that is AeroMedEvac

The tags when its on this mixer feeding IceCast should come out

462.950 = AirOps Pxxx 
123.025 = Air to Air
122.825 - AIROPS UNICOM
155.340 = AIR SECNDRY
155.2220 = Lifexxxx
853.3875 = xFD Talk Around

I also would have a feed like Region 7 MA on IceCast that covers

855.3875 = xFD TALK Around

So that tag should be sent to both feeds when it that frequency is active. 

Same as I might have the various 

123.025, 123.050, 123.075, 122.950  

123.025 = Air To Air
123.050 = Air to Air

So 123.025 when active would get sent to the AeroMedEvac mixer to that feed and that tag applied  to the above mixer feeding Kxxx Aviation and that tag sent there too...

Basically if the frequency has a tag, it should go out to whatever feed that it is sent to... be 1,2,3,x feeds/mixers

szpajder, 2018-11-22:
I prefer the current behaviour. I often listen to various streams while doing some work. When I hear something interesting, I may switch to the player window quickly to find out what was the frequency of the last call. And it works even if the chatter was very short. "Scanning..." does not convey any information - if the feed is silent, you know, it's scanning, don't you? It would just be an eye candy, occasionally obliterating the information which you are looking for.

szpajder, 2018-11-22:
@rec9140: and what should happen when two channels mixed by a common mixer start talking at the same time? Which tag would you like to see in your player? ("both" is not a correct answer - freq tag is emulated using Icecast song title property, and no song can have two titles at the same time).

ernieek, 2018-11-22:
Is it possible to make "scanning..." an option?

szpajder, 2018-11-22:
Does your life depend on it?

It is possible, however there are more important features to be implemented first.

ernieek, 2018-11-22:
okay.

But yes would be very nice in the situation i use it.

And it i think not a big code change i hope😊

rec9140, 2018-11-22:
>and what should happen when two channels mixed by a common mixer start talking at the same >time? Which tag would you like to see in your player? ("both" is not a correct answer - freq tag is >emulated using Icecast song title property, and no song can have two titles at the same time).

Thanks, I am aware of how the tags are sent. I use a BASH script and some python on others to update the MP3 metatag aka song title data as I grab it from the CC output..

BOTH IS the correct answer...

123.025 comes active.... so 123.025 audio comes out with its tag.. if say 462.950 is active as well you just slam the audio together? that is wrong.

Audio would be sequential to the mixer output.

123.025 audio/tag, 462.950 audio/tag....... 

see sdrtrunk... 

TG's might be active on 5 feeds, and each is fed with the audio and its tag... (albeit there is an issue with the tag feed right now), but the audio..

So you are just taking the audio and garbling it up if two frequencies are active????

szpajder, 2018-11-22:
Audio is sequential = it's a scanner.
Audio is simultaneous = it's a mixer.

These are two very different features of the program. Which one are you using actually?

closed 2018-12-17 by szpajder