rtl-airband/RTLSDR-Airband #107: Viewing Waterfall via Systemd Service Option

jlpapple, 2018-12-16:
Is there a method to display the waterfall when running as a service?

I've independently tried removing "-F", "-Fe" and changing "simple" to "forking" in the service file to no effect.  (I've run `sudo systemctl daemon-reload` after making the change.)

szpajder, 2018-12-16:
No, there isn't.

If you want to run the process in the background, then run it as a service. It has no terminal to print it's output then.

If you want the process to output data to the terminal, then run it in foreground, not as a service.

jlpapple, 2018-12-16:
Thanks.  

Normally I'd run it in the foreground, but I'm relying on the rtl_airband.service `Restart=always RestartSec=3` option to manage the random exit on fatal error as an [interim solution](https://github.com/szpajder/RTLSDR-Airband/issues/102#issuecomment-447667502).

The `Restart=always` option keeps [my public stream](https://www.broadcastify.com/listen/feed/29527) up as much as possible.

closed 2019-02-24 by szpajder