rtl-airband/RTLSDR-Airband #108: run a single or multiple instances?

danmilon, 2018-12-17:
Hey,

Not an issue, more like a question.
Is there a difference in performance if I run a separate instance of rtlsdr-airband for each rtlsdr dongle, instead of a single one for all of them?

Looking at the code, it seems that it runs a separate thread for each device, right? so it should be equivalent. What about GPU utilization?

Thanks!

szpajder, 2018-12-17:
I don't know, as I've never done any comparisons or measurements of such scenarios.

But on a Pi it's quite simple - the first instance starts and runs and remaining ones fail to start, as the GPU can't be used simultaneously by multiple processes.

danmilon, 2018-12-17:
Thanks for the prompt reply. Interesting, than that implies I haven't compiled (or am not running) properly for FFT on GPU, cause I got two of them to run. Will check what's up with that.

Thanks again.

closed 2018-12-17 by danmilon