rtl-airband/RTLSDR-Airband #109: reverberation/echo effect in mix of two channels

mhaberler, 2019-01-22:
have  a listen at 16:10 of http://static.mah.priv.at/public/LOWG1-Twr-App-Jan-22-2019-0730Z.mp3

there's echo - and I do not understand why, used to work just fine

note this is a mixer channel - the underlying Approach and Tower mp3's seem fine, those are mixed with balance -1.0 and 1.0 respectively:

http://static.mah.priv.at/public/LOWG1-Twr-Jan-22-2019-0730Z.mp3
http://static.mah.priv.at/public/LOWG1-App-Jan-22-2019-0730Z.mp3

platform: rpi2, cpu governor: performance
4 NESDR dongles, 4 channels in total
commit 6f9ffa

puzzled - any suggestions?

- Michael

mhaberler, 2019-01-22:
That was definitely the cause for the echo. 

closing.

closed 2019-01-22 by mhaberler

szpajder, 2019-01-24:
I have used the mixing feature many times, both on AM and NFM channels and I've never have this problem.

Did it show up right after you started the program? Or was it initially working fine and the effect popped up later?

Did a restart make it disappear or not?

mhaberler, 2019-01-25:
I had the station working fine several months, feeding liveatc
I decided to try some NFM reception and recompiled with NFM=1 but did not activate any NFM channels yet, and rebooted - however, did not check for reception quality thereafter and noticed the echo only some days later
reboot did definitely not cure it

if you are interested I can reproduce and give you access to the box, haberlerm at  gmail dot com

szpajder, 2019-01-25:
With four dongles on a RPi2 you are already at the edge of GPU performance
even with NFM support disabled. It has to calculate 4 x 8000 = 32000 FFT
transforms per second, which is close to its limit (at the default FFT size
of 512). Enabling NFM doubles the processing rate = you need 4 x 16000 =
64000 transforms per second. This won't work. You'll get choppy audio even
when not using mixers.

Please try the following:

- reenable NFM mode
- reduce sampling rate on all dongles by half (put "sample_rate = 1280000;"
in each device config section)
- reduce the FFT size by half ("fft_size = 256;" in the global config
section).

mhaberler, 2019-01-25:
Well I rather leave this box as "production" without NFM for liveatc - not that I really need NFM right there. I can experiment on a testbox though.

thanks for clarifying the performance issue!

mhaberler, 2019-01-25:
curious:
I had the sampling rate already reduced on some channels (1.4, 1.0, 1.0, and channel 4 disabled as I have dumpvdl2 running there)

is there a suggested "downscaling" ratio for samplerate like power of two?

what exactly  is the effect of reducing FFT size (besides reducing GPU needs) - more aliasing? narrower bandwidth? (sorry, not up to speed on digital radio)

szpajder, 2019-01-25:
Downscaling ratio is not very important. The program will produce a 8000
(or 16000) sps stream from any input sampling rate.

Lower FFT size = less FFT bins (channels) = wider channels. But if you
reduce FFT size by half and reduce the sampling rate by half at the same
time, then the effective channel width won't change.