rtl-airband/RTLSDR-Airband #110: Stream plays silence between transmissions using pulseaudio

barjac, 2019-02-01:
I use RPi running rtl_airband with pulse output.
The machine running the PA server also runs pulseeffects to filter the audio.
I am seeing very high cpu usage in pulseeffects between transmissions which I have reported to the pulseeffects developer here:
https://github.com/wwmm/pulseeffects/issues/423#issuecomment-459575012
It seems from his last comment that the stream should not  be playing silence between transmissions.
I will cross link to this bug report in that one so he will see your reply.
Thanks,
Barry

szpajder, 2019-02-01:
Try using "continuous = true;" option in the pulse output configuration.
Then it won't pause when squelch is closed.

barjac, 2019-02-01:
Is that valid for 2.4.0, as it makes no difference that I can see.
corked is still 'no' with either continuous = true; (or false as that seemed more logical so I tried it!).
I did re-boot between tests.
I will build 3.0.1 if that is likely to fix it, but current running version is 2.4.0.

szpajder, 2019-02-01:
2.4.0 and 3.0.1 work the same way. You have two options:

- continuous = false (the default) - when the squelch is closed,
rtl_airband will not be sending anything to the PA server. From the
server's perspective, there would be no input at all, so the stream will be
silent, because there is nothing to play.

- continuous = true - when the squelch is closed, rtl_airband will send
samples having a value of zero. The server will play silence, because a
continuous stream of zero-valued samples represents silence.

Choose the behavior which suits you better. To see the difference, you may
monitor the network traffic between rtl_airband and the server. With the
option set to true, you should see a continuous stream of traffic, and with
false there will be no traffic when the squelch closes.

barjac, 2019-02-01:
OK understood, I have updated to 3.0.1 anyway, and with continuous set to false I can see that there is traffic only during transmissions, and with it set true the traffic is continuous at the same level. (38kB/s)
However this has no effect on the high cpu usage experienced, so maybe this is unrelated.
The pulseeffects developer is expecting 'Corked: yes' to be set when the squelch is closed, and seems to think this should be controlled by rtl_airband. What are your thoughts on that?

szpajder, 2019-02-01:
I can't reproduce this problem. The CPU usage of pulseeffects process is the same, regardless of squelch state and continuous option setting. The exact value depends on the number of effects enabled, but it's no higher than 20% on a i5-6600K CPU. I checked both 8 ksps and 16 ksps stream (ie. NFM support disabled and enabled). Therefore I think the way rtl_airband drives the pulse stream is irrelevant.

Maybe it's related to a particular effect or effect combination? Or a specific version of pulseeffects, pulseaudio or some library? My setup:

- pulseeffects-4.4.6 (from Gentoo ebuild)
- pulseaudio-12.2
- calf-0.90.0
- mda-lv2-1.2.2
- fluidsynth-1.1.11
- gst-plugins-lv2-1.14.3
- gst-plugins-ladspa-1.14.3
- gst-plugins-bad-1.14.3
- gst-plugins-good-1.14.4
- gst-plugins-base-1.14.4
- gstreamer-1.14.4
- zita-convolver-3.1.0
- libsamplerate-0.1.9
- libsndfile-1.0.28

I tried auto gain, filter, equalizer, bass enhancer, crystalizer and reverberation. No issues.

wwmm, 2019-02-01:
If I am not mistaken @barjac is running both softwares in a raspberry. So I would expect a high cpu usage when audio processing is done with heavy effects or too many equalizer bands. The default number of 30 bands may be too much for a raspberry. In my opinion the only significant optimization that can be done here is not doing audio processing when the audio application (Airband in this case) is paused. 

In order to pause/unpause PulseEffects Gstreamer pipeline I need to know if there is at least one audio app playing audio. The only way I found to get this info from Pulseaudio was reading the corked property of Pulseaudio's `sink inputs` (audio app). `Corked = yes` means that the `sink input` stream is not running. If it is not corked the stream is running. Open a video or song in an app like `mpv` and you will see the corked state changing with every pause/unpause in the output of this command `pactl list sink-inputs`.

Based on what @szpajder explained above it seems to me that he is using `pa_stream_cork` to pause/unpause the stream. So everything should be fine. The corked property should be being updated. Could it be that this information is lost when you interact with Pulseaudio remotely?

wwmm, 2019-02-01:
Reading @barjac  email again it seems that only Airband is in the raspberry... In any case the cpu usage will reflect the selected effects. The general idea explained above remains...

wwmm, 2019-02-02:
I see. This explains why we are not able to pause the pipeline.

I agree. CPU usage should be the same playing silence. Unless there is a significant difference in buffer size when playing silence. At least the auto volume plugin consumes way more cpu when small buffers are used. This happens because `libebur128` functions will be called more often than they would be with larger buffers. And those functions are heavy. 

I don't think that buffer size should make much different to the other plugins. But I wouldn't rule out the possibility that something similar happens deep inside GStreamer. Maybe it has to send more buffers per second when they are small and this could lead to something eating more cpu.

wwmm, 2019-02-02:
Forget what I said. A difference in Airband buffer size would matter to pulseaudio but not to PulseEffects. Our buffer would remain the same unless some settings like buffer, latency and block size where changed...

barjac, 2019-02-02:
Having just looked back at the above it seems that instances of libgstequalizer start only when sound starts, but are not killed when it stops. Could this be the issue?

wwmm, 2019-02-02:
There are only two ways to stop the equalizer. Using the switch to turn it off or putting the whole pipeline in the paused state. But this last possibility relies on the corked property

barjac, 2019-02-02:
Comparison of installed packages:
`    pulseeffects-4.4.6 (from Gentoo ebuild)	pulseeffects-4.4.7
    pulseaudio-12.2				pulseaudio-12.2
    calf-0.90.0					calf-0.90.1
    mda-lv2-1.2.2			??	NOT INSTALLED
    fluidsynth-1.1.11					fluidsynth-2.0.3
    gst-plugins-lv2-1.14.3		??	NOT INSTALLED
    gst-plugins-ladspa-1.14.3			gstreamer1.0-ladspa-1.14.4
    gst-plugins-bad-1.14.3			gstreamer1.0-plugins-bad-1.14.4
    gst-plugins-good-1.14.4			gstreamer1.0-plugins-good-1.14.4
    gst-plugins-base-1.14.4			lib64gstreamer-plugins-base1.0_0-1.14.4
    gstreamer-1.14.4				lib64gstreamer1.0_0-1.14.4
    zita-convolver-3.1.0		???	lib64zita-alsa-pcmi0-0.3.2
    libsamplerate-0.1.9				lib64samplerate0-0.1.9
    libsndfile-1.0.28				lib64sndfile1-1.0.28`

barjac, 2019-02-02:
[pulseeffects_pkgs.txt](https://github.com/szpajder/RTLSDR-Airband/files/2824945/pulseeffects_pkgs.txt)

barjac, 2019-02-02:
There are 13 repeating calls to libgstequalizer.so that go on for ever after the first transmission is received. Adding up a random 13 they total 71%
[baz@localhost ~]$ perf top -p 11269 | grep libgstequalizer
    10.69%  libgstequalizer.so         [.] 0x00000000000025f1
    10.61%  libgstequalizer.so         [.] 0x00000000000025dc
    10.49%  libgstequalizer.so         [.] 0x00000000000025af
    10.23%  libgstequalizer.so         [.] 0x0000000000002602
     9.67%  libgstequalizer.so         [.] 0x00000000000025c8
     3.81%  libgstequalizer.so         [.] 0x0000000000002607
     3.63%  libgstequalizer.so         [.] 0x00000000000025f5
     3.61%  libgstequalizer.so         [.] 0x00000000000025cc
     3.33%  libgstequalizer.so         [.] 0x00000000000025e1
     2.59%  libgstequalizer.so         [.] 0x00000000000025b3
     0.95%  libgstequalizer.so         [.] 0x00000000000025c0
     0.75%  libgstequalizer.so         [.] 0x000000000000261e
     0.53%  libgstequalizer.so         [.] 0x00000000000025fa
     0.43%  libgstequalizer.so         [.] 0x0000000000002622
    10.78%  libgstequalizer.so         [.] 0x00000000000025dc
    10.45%  libgstequalizer.so         [.] 0x00000000000025f1
    10.32%  libgstequalizer.so         [.] 0x0000000000002602
    10.32%  libgstequalizer.so         [.] 0x00000000000025af
     9.70%  libgstequalizer.so         [.] 0x00000000000025c8
     3.68%  libgstequalizer.so         [.] 0x00000000000025cc
     3.65%  libgstequalizer.so         [.] 0x0000000000002607
     3.43%  libgstequalizer.so         [.] 0x00000000000025f5
     3.42%  libgstequalizer.so         [.] 0x00000000000025e1
     2.66%  libgstequalizer.so         [.] 0x00000000000025b3
     1.08%  libgstequalizer.so         [.] 0x00000000000025c0
     0.76%  libgstequalizer.so         [.] 0x000000000000261e
     0.49%  libgstequalizer.so         [.] 0x00000000000025fa
     0.48%  libgstequalizer.so         [.] 0x00000000000025a3
    10.54%  libgstequalizer.so         [.] 0x0000000000002602
    10.52%  libgstequalizer.so         [.] 0x00000000000025f1
    10.38%  libgstequalizer.so         [.] 0x00000000000025dc
    10.16%  libgstequalizer.so         [.] 0x00000000000025af
     9.89%  libgstequalizer.so         [.] 0x00000000000025c8
     3.69%  libgstequalizer.so         [.] 0x00000000000025cc
     3.65%  libgstequalizer.so         [.] 0x0000000000002607
     3.40%  libgstequalizer.so         [.] 0x00000000000025f5
     3.25%  libgstequalizer.so         [.] 0x00000000000025e1
     2.75%  libgstequalizer.so         [.] 0x00000000000025b3
     1.15%  libgstequalizer.so         [.] 0x00000000000025c0
     0.63%  libgstequalizer.so         [.] 0x000000000000261e
     0.55%  libgstequalizer.so         [.] 0x00000000000025d4
     0.44%  libgstequalizer.so         [.] 0x00000000000025fa
..........       --------------snip---------------

szpajder, 2019-02-03:
> I have run perf as requested and as far as I can tell it is libgstequalizer.so that is loading the cpu, however I see no correlation between the load shown in perf top and in top.

top shows percentages relative to the total CPU time available, while perf references them to the current CPU busy time.

> Using the PE GUI to switch off the equalizer causes the load to drop right down in top to ~5%, switching on the equalizer with no sound playing raises it to ~10% when a transmission comes along the load does not change, it is only when that first transmission finishes that the load increases to ~60%.

And just to be 100% sure: if you set "continuous = true" in the pulse output section in rtl_airband.conf, does the load also jump to ~60% after first transmission?

barjac, 2019-02-03:
Thanks for the explanation - I never had cause to use perf before.
> And just to be 100% sure: if you set "continuous = true" in the pulse output section in rtl_airband.conf, > does the load also jump to ~60% after first transmission?
Yes it make no noticeable difference.
I now have it set true and have re-started the rtl_airband service. (Also rebooted the RPi2 to be sure)
With the equalizer OFF pulseeffects runs at 0.3%
With PE ON and no transmission PE sits at 6%
During the first transmission it rises to ~8% 
When first transmission or any susbsequent ones stop it rises to ~70%
During any subsequent transmission it drops to ~8%

wwmm, 2019-02-03:
I did some tests with `gst-launch-1.0` and I think I found the source of the problem. Something odd happens when GStreamer is handling a buffer full of zero values. This command shows the behavior you are  seeing

`gst-launch-1.0 fakesrc format=3 sizetype=2 filltype=zero ! audio/x-raw,format=F32LE,channels=2,rate=44100 ! audioconvert ! equalizer-10bands band0=1.0 band1=1.0 band2=1.0 band3=1.0 band4=1.0 band5=1.0 band6=1.0 band7=1.0 band8=1.0 band9=1.0 ! pulsesink`

Now change the `filltype` to `random` instead of zeros and the output of perf changes

`gst-launch-1.0 fakesrc format=3 sizetype=2 filltype=random ! audio/x-raw,format=F32LE,channels=2,rate=44100 ! audioconvert ! equalizer-10bands band0=1.0 band1=1.0 band2=1.0 band3=1.0 band4=1.0 band5=1.0 band6=1.0 band7=1.0 band8=1.0 band9=1.0 ! pulsesink`

The equalizer is not the one with the highest cpu usage anymore. `libglib-2.0.so` nows consumes more cpu.

Considering that the only difference between the commands is if the buffer is filled with zeros or not I'd say that this is the cause of the difference in cpu usage. But I don't know yet why GStreamer is handling zero filled buffers this way. It may be worth ask questions in GStreamer mailing list.

barjac, 2019-02-03:
I think you need to re-start perf after starting/changing the streams to get reliable readings.
I just did the same tests and took these screen shots, but re-started perf after starting the streams in each case.
Zeros:
[image]
Random:
[image]

wwmm, 2019-02-03:
Yes I restarted perf in every test. And I also killed pulseeffects process `pulseeffects -q` before the tests so I could look only at GStreamer. Just give `gst-launch-1.0` pid to perf.

In any case your screenshots tell the same story. GStreamer's equalizer behaves differently when given a zero filled buffer. And not in a good way... It makes no sense...

barjac, 2019-02-04:
OK understood, I have repeated as you did and there is indeed something strange. I will put two screen shots here for reference in a gstreamer bug report.
random stream:
[image]
zero fill:
[image]

Would you like to report it or shall I? You can probably explain it better ;)

szpajder, 2019-02-04:
But this is nowhere near the 60% CPU usage which you get when equalizing rtl_airband's stream. And actually the allzero stream seems to cause lower CPU usage than random, which is exactly the opposite of what you are trying to show here. I guess the original bug hasn't been triggered in this demonstration.

wwmm, 2019-02-04:
In my computer both cases have the same low cpu usage in top. Only in perf there is any difference.

wwmm, 2019-02-04:
I am out of ideas. The only difference I see in his reports between the high and normal cpu usage are the zero filled buffers. But it doesn't matter what I try I can't get any difference in top. Even running PulseEffects and processing `gst-launch` output. Only perf shows any change... But this is not enough to open a bug report in GStreamer in my opinion

barjac, 2019-02-04:
I have just made a clean net installation of Mageia 7 from the current repo to rule out any issues with my rather old updated installation that I was using for test. It is even worse.
[image]
 I am now wondering whether we are missing some essential dependency, e.g. we don't currently have zita-convolver packaged:
Library zita-convolver found: NO
Message: Missing dependency zita-convolver = 3.x.x or zita-convolver = 4.x.x
Message: Convolver plugin will not be built.
Maybe you could look though the rest of the log for any clues?
http://mtf.duckdns.org/pub/linux/barjac/distrib/cauldron/x86_64/log/pulseeffects-4.4.7-3.mga7.src.rpm/build.0.20190131211017.log

wwmm, 2019-02-04:
I see nothing strange in your log. `zita-convolver` is used only by the convolver.

barjac, 2019-03-13:
There is now no issue with our most recent packaging that pulls in all the plugins. So closing. Thanks for your help.

closed 2019-03-13 by barjac