rtl-airband/RTLSDR-Airband #116: Odroid C2 - Failed to set up stream for SoapySDR device 'driver=sdrplay'

Strykar, 2019-03-13:
rtlsdr_airband works fine on my Odroid C2 with a local Elonics 4000 dongle, but not an RSP2Pro via Soapy.
SoapySDRremote and the RSP2 work fine with CubicSDR on the LAN -
```
SoapySDRServer[2501]: SoapyServerListener::accept([::ffff:192.168.15.165]:49708)
SoapySDRServer[2501]: SoapyServerListener::close()
```

I've snipped the serial number from rtl_airband logs, I've tried it without `hardware=` too -
```
rtl_airband[29935]: RTLSDR-Airband version v3.0.1-12-g9523dc9 starting
rtl_airband[29935]: Connecting to 127.0.0.1:8000/stream.mp3...
rtl_airband[29935]: Connected to 127.0.0.1:8000/stream.mp3
rtl_airband[29935]: SoapySDR: device 'driver=sdrplay,hardware=00000': sample rate set to 3000000 sps
rtl_airband[29935]: SoapySDR: device 'driver=sdrplay,hardware=1707065E10': antenna set to 'Antenna A'
rtl_airband[29935]: SoapySDR: device 'driver=sdrplay,hardware=00000': gain 'IFGR' set to 35.0 dB
rtl_airband[29935]: SoapySDR: device 'driver=sdrplay,hardware=00000': gain 'RFGR' set to 4.0 dB
rtl_airband[29935]: SoapySDR: device 'driver=sdrplay,hardware=00000' initialized
rtl_airband[29935]: Failed to set up stream for SoapySDR device 'driver=sdrplay,hardware=00000':
rtl_airband[29935]: 1 device(s) failed to initialize - aborting
```
rtl_airband's stable version logs even less -
`rtl_airband[32153]: Failed to open SoapySDR device 'driver=sdrplay,hardware=00000': map::at`

SoapySDR info -
```
$ SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=0000000
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=2
  remote:version=0.5.2-ga6fbf2db

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8)]
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * ExtRef Enable - External Reference Control
       [key=extref_ctrl, default=true, type=bool]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Stream args:
     * Remote Format - The stream format used on the remote device.
       [key=remote:format, default=CS16, type=string, options=(CS16, CF32)]
     * Remote Scale - The factor used to scale remote samples to full-scale floats.
       [key=remote:scale, default=32767.000000, type=float]
     * Remote MTU - The maximum datagram transfer size in bytes.
       [key=remote:mtu, units=bytes, default=1500, type=int]
     * Remote Window - The size of the kernel socket buffer in bytes.
       [key=remote:window, units=bytes, default=44040192, type=int]
     * Remote Priority - Specify the scheduling priority of the server forwarding threads.
       [key=remote:priority, default=0.500000, type=float, range=[-1, 1]]
     * Remote Protocol - Specify the transport protocol for the remote stream.
       [key=remote:prot, default=udp, type=string, options=(udp, tcp, none)]
  Antennas: Antenna A, Antenna B, Hi-Z
  Corrections: DC removal
  Full gain range: [0, 47] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 8] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.25, 0.5, 1, 2, 2.048, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz

$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-g4bfcbc74
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libremoteSupport.so  (0.5.2-a6fbf2d)
Module found: /usr/local/lib/SoapySDR/modules0.8/librtlsdrSupport.so  (0.3.1-c8c5b92)
Module found: /usr/local/lib/SoapySDR/modules0.8/libsdrPlaySupport.so (0.3.0-31a3a36)
Available factories... remote, rtlsdr, sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```
My airband conf -
```
$ cat /usr/local/etc/rtl_airband.conf
devices:
({
  type = "soapysdr";
  device_string = "driver=sdrplay,hardware=0000000000";
  gain = "IFGR=35,RFGR=4";
  correction = 0.0;
  antenna = "Antenna A";
  mode = "scan";
channels:
  (
    {
      freqs = ( 121.5, 122.8, 119.7 );
      labels = ( "Tower", "Ground", "Approach" );
      outputs: (
        {
	  type = "icecast";
	  server = "127.0.0.1";
          port = 8000;
          mountpoint = "stream.mp3";
          name = "Tower + Ground + Approach";
          genre = "ATC";
          description = "Local airport - aggregated feed";
          username = "source";
          password = "pass";
	  send_scan_freq_tags = true;
        },
        {
          type = "file";
          directory = "/home/strykar/recordings";
          filename_template = "TWR+GND+APP";
        }
      );
    }
  );
 }
);
```
What am I missing here?

szpajder, 2019-03-13:
Do you want to use a local SDRPlay device or a remote one, via SoapyRemote?

Strykar, 2019-03-13:
@szpajder Either would do TBH. In this case, Soapy is setup and working is all.

szpajder, 2019-03-13:
But in which scenario do you get the " Failed to open SoapySDR device "
error? When the device is local or remote?

You have configured a local device in rtl_airband.conf, but you've
mentioned SoapySDRServer too, so I'm confused.

If the device is local and you have SoapySDRServer running on the same
machine, then I suggest stopping it first, just to make sure that it
doesn't block the SDR for other apps.

Strykar, 2019-03-13:
rtl_airband and the rsp2 are local to the same machine - the Odroid C2.
I've never tried to connect to the RSP2 directly since I didn't see it supported outside of Soapy, hence the `type = "soapysdr"`
How would I connect rtl-airband to the rsp2 without soapy? I don' recall seeing that option for anything other than rtl-sdr on the wiki.
FWIW, I have tried `device_string = "remote=182.168.10.100,driver=remote,remote:timeout=100000,remote:driver=sdrplay";`
which also did not work.

Strykar, 2019-03-13:
Update: I stopped SoapySDRServer and tried rtl-airband again, same error.

szpajder, 2019-03-13:
So please post the output of  SoapySDRUtil --probe command now, with
SoapySDRServer stopped.

The output you've posted earlier only shows a remote device, not a local
one.

Strykar, 2019-03-13:
I'm not sure what gave you the impression that the device was remote in the previous `SoapySDRUtil --probe` output, but here you go -

```
$ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  driver = sdrplay
  label = SDRplay Dev0 RSP2 000000
  serial = 000000

$ SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=1707065E10
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=2

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8)]
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * ExtRef Enable - External Reference Control
       [key=extref_ctrl, default=true, type=bool]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: Antenna A, Antenna B, Hi-Z
  Corrections: DC removal
  Full gain range: [0, 47] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 8] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.25, 0.5, 1, 2, 2.048, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz
```

szpajder, 2019-03-13:
>
> I'm not sure what gave you the impression that the device was remote in
> the previous SoapySDRUtil --probe output, but here you go -
>
Now the RX Channel 0 section looks quite different, doesn't it?

The hardware serial number in the device identification sections looks real
now. So adjust your device_string in rtl_airband.conf and try again.

Strykar, 2019-03-13:
Same error -
```
$ /usr/local/bin/rtl_airband -fe -c /usr/local/etc/rtl_airband.conf
SoapySDR: device 'driver=sdrplay,hardware=1707065E10': using native sample format 'CS16' (fullScale=32767.0)
RTLSDR-Airband version v3.0.1-12-g9523dc9 starting
Connecting to 127.0.0.1:8000/stream.mp3...
Connected to 127.0.0.1:8000/stream.mp3
SoapySDR: device 'driver=sdrplay,hardware=1707065E10': sample rate set to 2048000 sps
SoapySDR: device 'driver=sdrplay,hardware=1707065E10': antenna set to 'Antenna A'
SoapySDR: device 'driver=sdrplay,hardware=1707065E10': gain 'IFGR' set to 35.0 dB
SoapySDR: device 'driver=sdrplay,hardware=1707065E10': gain 'RFGR' set to 4.0 dB
SoapySDR: device 'driver=sdrplay,hardware=1707065E10' initialized
[INFO] Using format CS16.
Failed to set up stream for SoapySDR device 'driver=sdrplay,hardware=1707065E10': 
1 device(s) failed to initialize - aborting

$ cat /usr/local/etc/rtl_airband.conf
devices:
({
  type = "soapysdr";
  device_string = "driver=sdrplay,hardware=1707065E10";
  gain = "IFGR=35,RFGR=4";
  correction = 0.0;
  sample_rate = 2.048;
  antenna = "Antenna A";
  mode = "scan";
channels:
  (
    {
      freqs = ( 121.5, 122.8, 119.7 );
      labels = ( "Tower", "Ground", "Approach" );
      outputs: (
        {
	  type = "icecast";
	  server = "127.0.0.1";
          port = 8000;
          mountpoint = "stream.mp3";
          name = "Tower + Ground + Approach";
          genre = "ATC";
          description = "Local airport - aggregated feed";
          username = "source";
          password = "rtlsdr";
	  send_scan_freq_tags = true;
        },
        {
          type = "file";
          directory = "/home/strykar/recordings";
          filename_template = "TWR+GND+APP";
        }
      );
    }
  );
 }
);
```

Strykar, 2019-03-13:
Is there a way to get more debugging info on where it's failing?
It appears to initialize -- (`SoapySDR: device 'driver=sdrplay,hardware=1707065E10' initialized`)
Then -- `Failed to set up stream for SoapySDR device 'driver=sdrplay,hardware=1707065E10':`

Strykar, 2019-03-13:
Strace log - https://gist.github.com/Strykar/a3773c94356eaab563b102f454d14fe2

szpajder, 2019-03-13:
The `Failed to set up stream...` message has a colon at the end and after that there should be an error string returned by SoapySDR library. If nothing shows up there, it means that SoapySDR returned an error code, but no error message.

There is nothing wrong with your config or with the program - I tested your configuration on my SDRPlay RSP1A and it works. So if there is no other application holding the device open, it should work.

szpajder, 2019-03-13:
My Soapy versions:

```
Lib Version: v0.7.0-g5f5a4d8c
API Version: v0.7.0
ABI Version: v0.7
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.7
Module found: /usr/local/lib/SoapySDR/modules0.7/libremoteSupport.so  (0.5.1-5a647db)
Module found: /usr/local/lib/SoapySDR/modules0.7/librtlsdrSupport.so  (0.3.0-f409a56)
Module found: /usr/local/lib/SoapySDR/modules0.7/libsdrPlaySupport.so (0.1.0-12c3db6)
Available factories... remote, rtlsdr, sdrplay
```

szpajder, 2019-03-13:
I know the cause - there was an API change in SoapySDR recently and it broke backwards compatibility. See here: https://github.com/pothosware/SoapySDR/issues/215. This is now fixed, so please update SoapySDR from the master branch, install it and recompile rtl_airband.

Strykar, 2019-03-13:
You sir are a champ, tyvm! Looking good in scan mode on the RSP2 now.

closed 2019-03-13 by Strykar