rtl-airband/RTLSDR-Airband #119: How to increasr recording quality?

mako19, 2019-03-18:
I see that MP3s generated are 24kBit/s, what do I need to change ans where to make it at least 128kBit/s, and generally as good as possible? I miss greatly the ausio filter that is present in SDRsharp. Without it my weak airband is not intelligible. Thanx.

szpajder, 2019-03-19:
>
> I see that MP3s generated are 24kBit/s, what do I need to change ans where
> to make it at least 128kBit/s, and generally as good as possible?
>
This is not configurable. You may modify the source code if you wish, but
it would just cause the 8ksps or 16ksps stream to be upsampled to 128 kbps,
so it won't improve the quality much.

> I miss greatly the ausio filter that is present in SDRsharp. Without it my
> weak airband is not intelligible. Thanx.
>
There is basic filtering support in unstable branch. See here for details:
https://github.com/szpajder/RTLSDR-Airband/pull/112

mako19, 2019-03-21:
Thanx. Where do I modify the code to make it 128 kbps? I have unstable brach so it is by default filtered 100-2500 Hz, and yet my receptions are not intelligible. But they are intelligible in SDRsharp. If only there was a workaround this!

szpajder, 2019-03-21:
>
> Thanx. Where do I modify the code to make it 128 kbps?
>

MP3 encoding parameters are set by airlame_init() function in output.cpp.
Variable bit rate is used by default, so if the codec decides that 8 or 16
kbps is enough at a particular time instant, it means it's enough and
bumping it up to any larger value won't make a significant difference.

> I have unstable brach so it is by default filtered 100-2500 Hz, and yet my
> receptions are not intelligible.
>

So don't waste your time with tweaking MP3 parameters. Your problem lies
elsewhere.
You can use Pulseaudio output to verify this. It plays uncompressed audio.

> But they are intelligible in SDRsharp. If only there was a workaround this!
>
Can you post an example recording from SDRSharp and rtl_airband, playing
the same channel (frequency) on the same receiver, the same antenna with
the same RTL gain and correction settings?

What is the noise level and signal level shown by SDRSharp and rtl_airband?

mako19, 2019-03-21:
Gain is index 14 which corresponds to 7.0dB according to rtl-test program.

mako19, 2019-03-21:
Here is a comparison between SDRSharp and Rtl-Airband with all parameters equal
https://dropmefiles.com/6MM4s
I had to change to strong frequency as rtl-airband records would be inaudible otherwize

szpajder, 2019-03-21:
The first thing I noticed is that the squelch does not close when there is
no transmission. Have you configured it manually to be always open? Is this
what you want?

Can you please run the program with -f option? It will then run in
foreground and display textual waterfall for each configured channel. The
number before "/" character is the signal level, the number after it is the
noise floor level estimate. What values do you get in these fields when the
channel is silent and when there is transmission?

Please post your rtl_airband.conf file (without passwords, etc).

mako19, 2019-03-21:
This is an inaudible recording example. Please note that no settings were changed on either dongle compared to the previous example, only frequency is now the weak one.
https://dropmefiles.com/PQW8Q
The two scrrenshots are made just before the recording and during the recording. Both Sharp and RTL-Airband are shown and a config file.
[image]

mako19, 2019-03-21:
Both of my dongles are FC0013
[image]

mako19, 2019-03-21:
Yes I prefer squelch always on, not to lose any data at the beginning of transmission. I don't care about file size.

danmilon, 2019-03-21:
* `ppm` correction is different. in sdr# it's 17, but in rtlsdr-airband it's 5.
Are you sure the gain in sdr# is 7db? I think `rtl-tcp` will report the value that sdr# asks it to set the gain to.

Also, what arguments do you pass to rtl-tcp?

mako19, 2019-03-21:
PPM values are different for the two dongles. See first screenshot. Dongles have serial numbers programmed. Dongle #3 is ppm=5 and dongle #4 is ppm=17.

No, I am not sure about gain. I assumed that marker position in SDRsharp corresponds to rtl-test value on that place. There are 24 values in rtl-test and 24 marker positions in SDRSharp. But I am not sure. Too sad Thomas eliminated auto gain. I used auto in older versions. It worked for me.

szpajder, 2019-03-21:
Have you tried higher gain values, like 18.8? Is it any better?

I don't have much experience with FC0013, so I can't tell how well it
performs with rtl_airband. I use R820Ts which is reportedly more sensitive
than FC0013. And I usually run them with gains around 30-38 dB (on a scale
from 0 to 49.6 dB), without external amplifiers.

Bear in mind though that this program is not really designed for DXing. Its
channelizer works on a different principle than the one in SDRSharp. It is
designed to process many channels simultaneously, at an expense of lower
selectivity. But in typical scenarios it performs well enough.

mako19, 2019-03-21:
What puzzles me is that with strong signal RTL-Airband works more-or-less, but on weak signal it simply loses everything. Is it gain problem or lack audio noise filter?

I will now try high gain. But I choose gain value using waterfall and SNR indicator in Sharp. It must be optimal.

mako19, 2019-03-21:
Tried 18.1 but it just as bad

mako19, 2019-03-21:
Tested it with E4000 but still no reception at weak signals only strong. Strange.

danmilon, 2019-03-21:
You run rtl_tcp and connect sdr# to it, correct? What arguments do you pass to rtl_tcp exactly?

szpajder, 2019-03-21:
In 121.8.mp3 the speech is not only weak, but also garbled. It's like the
radio is mistuned. Are you sure the correction is 5 and not, for example,
-5? Try to fiddle with this setting to find out if it makes any change.

Also it's usually not a good idea to set centerfreq to the same value as
channel frequency. Give it a shift of 50 or 100 kHz. Either side.

mako19, 2019-03-22:
Can you guys make a similar comparison? Sharp vs RTL-Airband on weak signals? Is it just me or it should be that way?

szpajder, 2019-03-23:
[image]

The stronger transmissions are from aircraft. The weak one is the tower controller (from ground transmitter).

SDRSharp:

https://drive.google.com/open?id=1xfvSkJxK2nwzTbmkrWYyH_7UcgOF4te-

rtl_airband:

https://drive.google.com/open?id=1rDtmObFH81rtGeoGeMUwhHhy84TS9P5d

Same antenna, same RTL dongle, so different points in time, but the ground transmitter power stays constant, so it's quite comparable.

rtl-sdr.com v3 dongle, end-fed dipole located ca 40 meters above terrain, no amplifiers, no filters. Airport transmitter located about 25 km away, mostly flat terrain with some obstacles (low buildings, etc).

rtl_airband settings:

- RTL gain 34 dB
- default sampling rate
- default FFT size
- default squelch settings
- centerfreq offset from channel frequency: 75 kHz
- waterfall shows 8/8 when the channel is silent, 35/8 when the ATCO is transmitting.

To get the recording in SDRSharp, I used a spyserver. SDRSharp gain was set to 15 (on a scale from 0 to 29 - I don't know how it's related to RTL gain which goes from 0 to 49.6 dB).

To me both recordings are intelligible most of the time. Indeed, audio sampling rate in SDRSharp is higher, but this mostly adds high frequency noise, not any extra sound quality.

mako19, 2019-03-23:
Thanx Tomas! Quality is very good indeed. This is what I expect from my setup but don't get for some strange reason... I'm using RTL-Airband since 2016 and it always used to be like your recordings. But recently something broke. I must reinstall it. Maybe install was botched? I simply deleted RTL-Airband folder and went through unstable branch step-by-step.

mako19, 2019-03-23:
I live under huge TV tower in Moscow. I absolutely need filter and I have a good one. And a Yagi custom made for me. All high quality equipment. But it does not explain the difference. It must be some software bug somewhere in y system.

mako19, 2019-03-23:
Maybe it's an audio AGC issue? The noise in my recordings is deafening. I had to use audio software to reduce volume. Was too loud.

szpajder, 2019-03-23:
The noise is deafening because you have very low signal to noise ratio and you have disabled squelch which causes the AGC to operate constantly, even when the squelch would normally be closed. So when there is no transmission, the noise gets amplified to a reasonably loud level for your listening comfort :)

I'm a bit puzzled with the distortion in your audio. Can you take a brief recording of baseband I/Q data when this occurs and when the audio is unintelligible? You can do this using "rawfile" output type (https://github.com/szpajder/RTLSDR-Airband/wiki/Configuring-rawfile-outputs).

mako19, 2019-03-24:
I deleted my RTL-Airband folder, deleted /usr/bin/rtl-airband file and installed stable 3.0.1 version. It seems the problem is solved.
https://dropmefiles.com/kAYsR

mako19, 2019-03-24:
First part is SDRSharp as always. Well, this is a good result but my question still applies - maybe we can increase a bitrate so we can apply filtering post-facto? Or it will not work with compressed formats? I am talking of filtering recorded files, not a stream. I tried to use Audacity noise filter function, but it had no effect. SDRSharp is still somewhat more intelligible. With weak signals every dB is precious.

szpajder, 2019-03-24:
> I deleted my RTL-Airband folder, deleted /usr/bin/rtl-airband file and installed stable 3.0.1 version. It seems the problem is solved.

The only change since 3.0.1 to current unstable which affects the sound quality is the introduction of filters, which are on by default, but you can turn them off.

> maybe we can increase a bitrate so we can apply filtering post-facto?

As I said before - the bitrate is dynamically adjusted by the MP3 VBR codec. If it thinks it's enough to convey the instantaneous bandwidth of the input signal, then it's enough and increasing the bitrate will not help at all.

Regarding https://dropmefiles.com/kAYsR - in my opinion both recordings are good enough (for the SNR you are able to get). rtl_airband audio is a bit louder and hence easier to listen. SDRSharp audio has a higher sampling rate, which gives more hiss, but contributes nothing to the overall speech quality. If you want even better quality, then what you need is higher SNR. Neither higher sampling rate nor higher MP3 bitrate nor super-duper DSP algorithms will give you this.

mako19, 2019-05-20:
Great program again, but I still see the audible difference betweein it and Sharp, which is critical for intelligibility of my receptions. Is there a way to save an undecoded I/Q stream, so that I can feed it to Sharp later? Saving it with rtl_tcp needs a lot of space. We onlyy need 10kHz really. Would be great.

szpajder, 2019-05-20:
https://github.com/szpajder/RTLSDR-Airband/wiki/Configuring-rawfile-outputs

mako19, 2019-05-20:
Yes I know but I couldn't feed it to Sharp, which is what I want to do.

mako19, 2019-05-22:
I'll ask it here...how importaint is sample loss at high rates? Do I understand it right that for our purpose it is negligible even at 3.2MSPS? The picture is 10 seconds of testing.
[image]

mako19, 2019-05-23:
No, I am wrong. 3,2MSPS doesn't work.

closed 2019-08-25 by szpajder