rtl-airband/RTLSDR-Airband #121: Scan mode has constant noise, issue with squelch?

Strykar, 2019-04-10:
When I originally setup scan mode, it was quiet on the Icecast stream unless there was traffic. Now I get a constant stream of noise instead. Have I mucked something up, I didn't make any changes IIRC?
My rtl_airband.conf:

```
$ cat /usr/local/etc/rtl_airband.conf
fft_size = 1024;
devices:
({
  type = "soapysdr";
  index = 7;
  device_string = "driver=sdrplay,hardware=1707065E10,rfgain_sel=5,if_mode=Zero-IF,iqcorr_ctrl=true,extref_ctrl=false,biasT_ctrl=false,rfnotch_ctrl=yes";
  gain = "IFGR=25,RFGR=3";
  correction = 0.0;
  sample_rate = 2.048;
  antenna = "Antenna A";
  mode = "scan";
channels:
  (
    {
      freqs = ( 242.0, 122.8, 121.1, 122.7, 119.1 );
      lowpass = 2500;
      highpass = 150;
      modulation = "am";
      afc = 0;
      labels = ( "MAD", "Tower", "IAD", "Ground", "Approach" );
      outputs: (
        {
	  type = "icecast";
	  server = "127.0.0.1";
          port = 8000;
          mountpoint = "stream.mp3";
          name = "MAD + Tower + IAD + Ground + Approach";
          genre = "ATC";
          description = "Air - aggregated feed";
          username = "source";
          password = "123456";
	  send_scan_freq_tags = true;
        },
        {
          type = "file";
          directory = "/home/strykar/recordings";
          filename_template = "MAD+TWR+IAD+GND+APP";
        }
      );
    }
  );
 }
);
```

vitaliy-sk, 2019-04-10:
What you will see if you run with -f option?
`rtl_airband -f`

szpajder, 2019-04-10:
Maybe there is some interference on one of the scanned frequencies and it blocks the scanner by keeping the squelch open constantly?

As @vitaliy-sk suggested - run the program with -f option, identify the frequency which causes the problem, remove it from the scanned frequencies list and verify if it helps.

Strykar, 2019-04-10:
Ah I see, it seems to be stuck on only 242 MHz.
Removing it seems to have fixed the squelch/noise, I'll keep this issue open while I monitor for a few days.
```
$ rtl_airband -f
[INFO] Using format CS16.

 242.000                                                                       

 409/100* 242.000
 408/100* 242.000
 410/100* 242.000
 411/100* 242.000
 405/100* 242.000
 402/100* 242.000
 405/100* 242.000
 422/100* 242.000
 399/100* 242.000^C
```
I take it there's no workaround short of removing the offending frequency from the list?

vitaliy-sk, 2019-04-11:
@Strykar you can try to set manual squelch for 242.000

smth like this 

```
freqs = ( 242.0, 122.8, 121.1, 122.7, 119.1 );
squelch = ( XXX, -1, -1, -1, -1 );
```

closed 2019-08-25 by szpajder