rtl-airband/RTLSDR-Airband #123: Can't compile with SoapySDR enabled

SJO-8164, 2019-04-18:
I am trying to compile for a Raspberry Pi 3, with NFM and SoapySDR support enabled. When I try to compile with the following command:

`make PLATFORM=rpiv2 NFM=1 WITH_SOAPYSDR=1`

The compilation will fail at this point:
```
make[1]: Leaving directory '/home/pi/rtlsdr-airband/RTLSDR-Airband/hello_fft' 
as -o rtl_airband_neon.o rtl_airband_neon.s
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.0.1-14-gca642be\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv7-a -mfpu=neon-vfpv4 -mfloat-abi=hard -ffast-math  -DNFM -DWITH_RTLSDR -DWITH_SOAPYSDR   -c -o input-rtlsdr.o input-rtlsdr.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.0.1-14-gca642be\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv7-a -mfpu=neon-vfpv4 -mfloat-abi=hard -ffast-math  -DNFM -DWITH_RTLSDR -DWITH_SOAPYSDR   -c -o input-soapysdr.o input-soapysdr.cpp
g++ -rdynamic -L/opt/vc/lib  rtl_airband.o input-common.o input-helpers.o output.o config.o util.o mixer.o hello_fft/hello_fft.a rtl_airband_neon.o input-rtlsdr.o input-soapysdr.o  -lrt -lm -ldl -lvorbisenc -lmp3lame -lshout -lpthread -lconfig++ -lbcm_host -ldl -lrtlsdr -lSoapySDR -o rtl_airband
input-soapysdr.o: In function `soapysdr_init(input_t*)':
/home/pi/rtlsdr-airband/RTLSDR-Airband/input-soapysdr.cpp:255: undefined reference to `SoapySDRDevice_setFrequencyCorrection'
input-soapysdr.o: In function `soapysdr_parse_config(input_t*, libconfig::Setting&)':
/home/pi/rtlsdr-airband/RTLSDR-Airband/input-soapysdr.cpp:168: undefined reference to `SoapySDRKwargs_fromString'
input-soapysdr.o: In function `sdrplay_get_nearest_sample_rate':
/home/pi/rtlsdr-airband/RTLSDR-Airband/input-soapysdr.cpp:116: undefined reference to `SoapySDRDevice_getSampleRateRange'
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'rtl_airband' failed
make: *** [rtl_airband] Error 1
```

Here is the output of `SoapySDRUtil --info`
```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-g9a942698
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libsdrPlaySupport.so (0.3.0-4e0c67d)
Available factories... sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

```
This will happen when trying to compile on both the stable and unstable branch. Compilation is successful without SoapySDR support enabled.

szpajder, 2019-04-18:
No problem here, using exactly the same version (commit ID) of rtl_airband
and SoapySDR.

Don't you have some old version of libSoapySDR.so laying around in
/usr/local/lib, etc?

SJO-8164, 2019-04-18:
Ah yes - you were correct. I found some old libSoapySDR.so files and removed them, then RTLSDR-Airband compiled successfully.

Thank you

closed 2019-04-18 by SJO-8164