rtl-airband/RTLSDR-Airband #124: Crashing/freeze on Raspberry Zero W - kernal error?

villevarn, 2019-04-21:
Hi

Installed 3.0.1 on a Rasperry Pi Zero W using the guide and compiled using make PLATFORM=rpiv1.
Using the default config file to stream to icecast2 server.

After 5-10 minutes the pi would freeze and drop the network. I googled away but ended up reinstalling Raspbian and start over. 

But the problem still occurs. The application will run but after approx. 5 minutes of runtime I get these messages before the SSH session gets unresponsive: 

 119.500   120.225

   6/  6     7/  6
   6/  6     7/  6
   7/  6     6/  6
Message from syslogd@raspberrypi at Apr 21 21:41:44 ...
 kernel:[ 1633.597044] Internal error: Oops: 837 [#1] ARM
   7/  6     6/  6
Message from syslogd@raspberrypi at Apr 21 21:41:44 ...
 kernel:[ 1633.597350] Process rtl_airband (pid: 2145, stack limit = 0xd7884188)
   7/  6     7/  6
Message from syslogd@raspberrypi at Apr 21 21:41:44 ...
 kernel:[ 1633.597360] Stack: (0xd7885fa8 to 0xd7886000)
   7/  6     6/  6
Message from syslogd@raspberrypi at Apr 21 21:41:44 ...
 kernel:[ 1633.597375] 5fa0:                   00169b00 000003e8 0017d3b8 00000003 000003e8 9cb15800

Message from syslogd@raspberrypi at Apr 21 21:41:44 ...
 kernel:[ 1633.597390] 5fc0: 00169b00 000003e8 00169c08 000000a8 b695d168 b695d6d4 00000000 b6970000

Message from syslogd@raspberrypi at Apr 21 21:41:44 ...
 kernel:[ 1633.597405] 5fe0: 00000000 b58aec30 00000000 b6b28124 80000010 0017d3b8 00000000 00000000

Message from syslogd@raspberrypi at Apr 21 21:41:44 ...
 kernel:[ 1633.597426] Code: f10c0080 e5992008 e35204bf 1b000de1 (e5991000)

This time the network does not drop yet and I can open a new SSH session to the pi. 

Any suggestions how I can get the application running on my rpi zero w?

szpajder, 2019-04-21:
After 5-10 minutes = ie. after it heats up?

First of all, please verify if it does not overheat and if the power supply is able to keep up with the load. Try another power supply and USB cables. See also https://github.com/szpajder/RTLSDR-Airband/issues/94.

villevarn, 2019-04-21:
It stays around 50-53 degrees C when running. Is that "overheating"? 
Power Supply is original 2.5A but I'll see to find another to test with & other cables.

szpajder, 2019-04-22:
50-53 deg seems OK.

No power supply can be characterized just with a single number (current rating). These are cheap devices and their behaviour may vary under load and with the PSU's age. Cables also make a difference (https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=88392#p621837)

It's a good idea to use a PSU with an adjustable output voltage (say, up to 6 volts). A 5V-rated PSU may give you 5 volts on its terminals, but when you add voltage drop on cables and connections and run a CPU-intensive task on the Pi (https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=61925&p=462502#p462392), it may fall below by half a volt or so and the Pi will no longer work reliably.

villevarn, 2019-04-22:
Hi 

Thanks for the quick replies. 
I only had another PSU from a Pi3 lying around. Tried that with another USB cable and same error after ~10 minutes. Also tried supplying power via USB 3.0 port from my PC but same results. About 10 min runtime then crash. 

I guess I'll try to find a third PSU, perhaps a unit from a PC and put some USB cables on the 5v line. It should be able to supply the power without drops?

szpajder, 2019-04-22:
And how about running rtl_airband on that Pi3? Does it fail the same way? (Make sure to bring the OS to the same state on both Pis with "apt-get upgrade" before jumping into conclusions).

I tried to power two RPi3s and an Odroid XU4 from an ATX PSU recently. It didn't work, because the PSU was giving exactly 5V on its output and after passing through all the cables and a power distribution panel it dropped down to about 4.7 under load and neither the RPis nor the Odroid even booted. I need a PSU cranked up to about 5.9V for this setup to work correectly. But it's three computers and five SDRs in total.

In general, RPi Zero is quite similar hardware-wise to RPi version 1 on which rtl_airband runs fine. Sure, it's always possible that some backward-incompatible change went into Raspbian recently, but if this was the case, I guess we would have more people here with complaints. And the program would probably crash straight away after starting it up, not after 10 minutes.

vitaliy-sk, 2019-04-22:
@szpajder do you have smth around this time in `/var/log/syslog`?

villevarn, 2019-04-24:
The Pi3 is busy doing other stuff so don't want to mess with it. :-) 
I'm still on the hunt for a better PSU with adjustable voltage but it might be a week or two until I can get my hands on one. I'll let you know if it works. And I agree that more people would probably have raised the issue if it was with the software.

@vitaliy-sk 
/var/log/syslog is showing:

Apr 24 22:50:54 raspberrypi rtl_airband[1042]: RTLSDR-Airband version 3.0.1 starting
Apr 24 22:50:54 raspberrypi rtl_airband[1042]: Connecting to 192.168.0.182:8000/TWR.mp3...
Apr 24 22:50:54 raspberrypi rtl_airband[1042]: Connected to 192.168.0.182:8000/TWR.mp3
Apr 24 22:50:54 raspberrypi rtl_airband[1042]: Connecting to 192.168.0.182:8000/GND.mp3...
Apr 24 22:50:54 raspberrypi rtl_airband[1042]: Connected to 192.168.0.182:8000/GND.mp3
Apr 24 22:50:55 raspberrypi rtl_airband[1042]: Device #0: gain set to 25.40 dB
Apr 24 22:50:55 raspberrypi rtl_airband[1042]: RTLSDR device 0 initialized
Apr 24 22:55:01 raspberrypi CRON[1063]: (pi) CMD (/usr/bin/sudo -H /usr/local/bin/checkwifi.sh >> /dev/null 2>&1)
Apr 24 23:00:01 raspberrypi CRON[1082]: (pi) CMD (/usr/bin/sudo -H /usr/local/bin/checkwifi.sh >> /dev/null 2>&1)
Apr 24 23:05:01 raspberrypi CRON[1093]: (pi) CMD (/usr/bin/sudo -H /usr/local/bin/checkwifi.sh >> /dev/null 2>&1)
Apr 24 23:05:20 raspberrypi kernel: [ 2377.644838] Unable to handle kernel paging request at virtual address ffffff68
Apr 24 23:05:20 raspberrypi kernel: [ 2377.644883] pgd = d8198000
Apr 24 23:05:20 raspberrypi kernel: [ 2377.644924] [ffffff68] *pgd=1bffa861, *pte=00000000, *ppte=00000000
Apr 24 23:05:20 raspberrypi kernel: [ 2377.644959] Internal error: Oops: 837 [#2] ARM
Apr 24 23:05:20 raspberrypi kernel: [ 2377.644975] Modules linked in: fuse rfcomm cmac bnep evdev hci_uart btbcm serdev bluetooth ecdh_g$
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645115] CPU: 0 PID: 1048 Comm: rtl_airband Tainted: G      D  C      4.14.98+ #1200
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645128] Hardware name: BCM2835
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645145] task: dad48000 task.stack: d63f4000
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645186] PC is at ret_fast_syscall+0x14/0x28
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645200] LR is at 0x0
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645212] pc : [<c000fd94>]    lr : [<00000000>]    psr: 60000013
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645222] sp : d63f5fa8  ip : d63f5c0c  fp : 00000000
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645233] r10: 00000000  r9 : d63f4000  r8 : c000ff24
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645244] r7 : 000000a8  r6 : 001f7c08  r5 : 000003e8  r4 : 001f7b00
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645256] r3 : 00000000  r2 : 00000001  r1 : 0020b3cf  r0 : 00000001
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645269] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645280] Control: 00c5387d  Table: 18198008  DAC: 00000055
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645269] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645280] Control: 00c5387d  Table: 18198008  DAC: 00000055
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645293] Process rtl_airband (pid: 1048, stack limit = 0xd63f4188)
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645306] Stack: (0xd63f5fa8 to 0xd63f6000)
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645323] 5fa0:                   001f7b00 000003e8 0020b3b8 00000003 000003e8 ea88bf00
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645341] 5fc0: 001f7b00 000003e8 001f7c08 000000a8 b68c3168 b68c36d4 00000000 b68d6000
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645359] 5fe0: 00000000 b5814c30 00000000 b6a8e124 80000010 0020b3b8 00000000 00000000
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645390] Code: f10c0080 e5992008 e35204bf 1b000de1 (e5991000)
Apr 24 23:05:20 raspberrypi kernel: [ 2377.645458] ---[ end trace 6fbea86460a6533e ]---
Apr 24 23:10:01 raspberrypi CRON[1104]: (pi) CMD (/usr/bin/sudo -H /usr/local/bin/checkwifi.sh >> /dev/null 2>&1)
Apr 24 23:15:01 raspberrypi CRON[1116]: (pi) CMD (/usr/bin/sudo -H /usr/local/bin/checkwifi.sh >> /dev/null 2>&1)
Apr 24 23:17:01 raspberrypi CRON[1129]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Apr 24 23:18:10 raspberrypi rtl_airband[1042]: Got signal 1, exiting
Apr 24 23:18:10 raspberrypi rtl_airband[1042]: Freeing GPU memory
Apr 24 23:18:10 raspberrypi rtl_airband[1042]: Cleaning up
Apr 24 23:18:10 raspberrypi rtl_airband[1042]: Input threads closed

villevarn, 2019-05-04:
Hi,

So I finally got my hands on another PSU. An old ATX supply. I connected a USB cable to the 5V supply and the multimeter states 5.30V. It drops to 5.24 with the pi booting up (with dongle on the USB port).
Now I got a different result. No dongle visible when typing lsusb. Connected a monitor. When booting I got two lines both stating:
usb usb1-port1: Cannot enable. Maybe the USB cable is bad?

Well, cable is fine. I took the dongle with the same USB cable and used it on my PC. Works fine. Even tried other cables (to connect dongle and also for power). Same error.
Seems like this must be some kind of power issue as you already recognized in the first post so I'll close this issue. 
Thanks a bunch for pointing me in the direction to find the problem.. still working on a solution. Will get my hands on a adjustable voltage supply and/or get the same dongle & cables tested on another pi.

closed 2019-05-04 by villevarn

szpajder, 2019-05-04:
Measuring the voltage directly on the PSU output terminals won't tell how much is lost on the microUSB connector pins on the RPi and on the traces of the RPi PCB. You should measure the voltage between +5V and GND pins on the GPIO head of the Pi (see https://pinout.xyz/ for GPIO pinout).

It might also be worth checking the voltage on the USB port of the Pi (or even at the end of the USB cable which you use to connect the dongle). If the USB device is not powered properly, it will behave erratically or won't be detected at all.