rtl-airband/RTLSDR-Airband #129: Feature Request: scanning metadata tag

redrob, 2019-06-18:
For scan mode metadata, it would be great to see a 'Scanning...' metadata tag for when RTLSDR-Airband is actively scanning and not stopped on a channel. This is a common, editable feature in other icecast/shoutcast clients which support dynamic scanner metadata. 

Currently, icecast will continue to show the last stopped on channel/alpha tag while scanning until it stops on a new channel.

Much appreciated.

dupilax, 2019-06-18:
By that you mean it could be possible to see on Icecast wich frequency is active ?!

szpajder, 2019-06-18:
This has already been requested. I don't think it's a good idea, see
https://github.com/szpajder/RTLSDR-Airband/issues/104#issuecomment-441102090

closed 2019-08-25 by szpajder