rtl-airband/RTLSDR-Airband #130: (code=exited, status=255/EXCEPTION)

herminiofer, 2019-07-13:
Good night

i tried to install  the RTLSDR-Airband, but i dont can it to work

pi@raspberrypi:/usr/local/bin $ sudo systemctl status rtl_airband                                                               ● rtl_airband.service - SDR AM/NFM demodulator
   Loaded: loaded (/etc/systemd/system/rtl_airband.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2019-07-13 17:25:43 BST; 6min ago
     Docs: https://github.com/szpajder/RTLSDR-Airband/wiki
 Main PID: 1266 (code=exited, status=255/EXCEPTION)

Jul 13 17:25:43 raspberrypi sudo[1266]: Allocating 10 zero-copy buffers
Jul 13 17:25:43 raspberrypi sudo[1266]: Detected Kernel usbfs mmap() bug, falling back to buffers in userspace
Jul 13 17:25:43 raspberrypi sudo[1266]: Connecting to audio.rb24.com:8000/EXTVHF000124...
Jul 13 17:25:43 raspberrypi sudo[1266]: Connected to audio.rb24.com:8000/EXTVHF000124
Jul 13 17:25:43 raspberrypi sudo[1266]: Device #0: gain set to 25.40 dB
Jul 13 17:25:43 raspberrypi sudo[1266]: RTLSDR device 0 initialized
Jul 13 17:25:43 raspberrypi sudo[1266]: mapmem(): mmap error: Cannot allocate memory
Jul 13 17:25:43 raspberrypi sudo[1266]: pam_unix(sudo:session): session closed for user root
Jul 13 17:25:43 raspberrypi systemd[1]: rtl_airband.service: Main process exited, code=exited, status=255/EXCEPTION
Jul 13 17:25:43 raspberrypi systemd[1]: rtl_airband.service: Failed with result 'exit-code'.

My system is:
Raspberry 4
pi@raspberrypi:/usr/local/bin $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

Can you help me solve these problems?

Thanks in advance

szpajder, 2019-07-13:
> mapmem(): mmap error: Cannot allocate memory

This error comes from the GPU-FFT code. Probably RPi4 has some backward incompatible GPU related changes which prevent this code from working. But I don't currently have RPi4 at hand, so can't tell for sure. In the meantime you can recompile with `PLATFORM-armv8-generic` or `armv7-generic` and it should work (but it won't utilize the GPU chip then).

herminiofer, 2019-07-13:
I will try tomorrow.
Then I give feedback
Thanks

herminiofer, 2019-07-14:
Good afternoon

I tried compiling for armv8-generic, but it gives the following error:
pi @ raspberrypi: /RTLSDR-Airband-3.0.1 $ sudo make PLATFORM = armv8-generic
g ++ -O3 -g -Wall -DSYSCONFDIR = \ "/ usr / local / etc \" -DDEBUG = 0 -march = armv8 -a + crc -mtune = cortex-a53 -ffast-math -DWITH_RTLSDR -c -o rtl_airband. rtl_airband.cpp
rtl_airband.cpp: 33: 10: fatal error: fftw3.h: No such file or directory
  #include <fftw3.h>
           ^ ~~~~~~~~
compilation terminated.
make: *** [<builtin>: rtl_airband.o] Error 1

danmilon, 2019-07-14:
You're missing dependencies. See wiki page [Dependencies](https://github.com/szpajder/RTLSDR-Airband/wiki/Dependencies).

herminiofer, 2019-07-14:
Good afternoon

Definitely can not for it to work
I'll try it on raspberry 3
Thanks

herminiofer, 2019-07-14:
Good afternoon

Finally I got for this to work

closed 2019-07-28 by herminiofer

farolone, 2020-10-27:
sudo apt install  libfftw3-dev
sudo apt install libconfig++-dev
sudo apt install libogg-dev libvorbis-dev libshout3-dev libmp3lame-dev

fixed it for me.