rtl-airband/RTLSDR-Airband #131: Compatibility question!

SA7BNT, 2019-07-14:
Hallo,
will the Pluto run with RTLSDR-Airband? 
Up to 20 MHz of instantaneous bandwidth should be nice ;)
[https://www.analog.com/en/design-center/evaluation-hardware-and-software/evaluation-boards-kits/adalm-pluto.html#eb-overview](url)

Best
Tim

szpajder, 2019-07-14:
There is a SoapySDR driver for it (https://github.com/pothosware/SoapyPlutoSDR). If it works, then rtl_airband should work as well. Be a pioneer - try it out and share your experience :)

SA7BNT, 2019-07-14:
I will order one and let you know about the result.
Thanks for info.

starvald, 2019-07-14:
Internally it may be able to receive 20MHz but trying from personal experience it isn't much better than an rtl-sdr, pushing that much data through the usb2 interface it can't handle it without dropping samples.

SA7BNT, 2019-07-14:
Well absolut right. Via USB no chance, but what with Network? Why are they making so manny nice hacks? ;)
https://wiki.analog.com/university/tools/pluto/users/customizing
https://hackaday.com/2019/05/10/pluto-sdr-goes-ethernet/

closed 2019-07-19 by SA7BNT