rtl-airband/RTLSDR-Airband #132: Pulseaudio streaming format and bandwidth

sm5gxq, 2019-07-30:
Hi,

Is there a way to use Pulseaudio to stream between a remote SDR and a central server? In that case, can I set the stream format parameters to something else than 32 bit, 44kHz? Eg aLaw (8kHz, 8bit), which seems to be supported by Pulseaudio?

Typically, the remote SDR would use a RPi and a mobile internet Connection, where bandwidth is limited (and costly).

rgds
Peter

szpajder, 2019-07-30:
Hi Peter,

Pulseaudio is more suitable for streaming over a local network, where
bandwidth is not a concern. In other cases just use icecast output - it
uses MP3 compression.

sm5gxq, 2019-07-30:
Hi,

thanks for the reply.

Yes I know there is Icecast (I used it many yearrs ago, but then only for one channel at a time), but when using multiple RX at the same frequency, Icecast is not really practically usable due to its unpredictable latency.

aLaw would be quite ok as "compression" in my case. So the question is, if it's possible to use this audio format as a Pulseaudio link between two computers over Internet. I was looking at the source code for Pulseaudio in RTL-airband and noticed that 32-bit float is hard coded. Is there a negotiation between the Pulseaudio server and RTL-airband to setup the stream parameters? 

Can RTL-airband send a stream as aLaw or some other format, less bandwidth-consuming than 44kHz/32bitFLOAT? Otherwise, perhaps it would be possible to stream with the default format to a local PA-server in the RPi and there, through a loopback interface, send it over the net to Another computer, but using a different streaming format and codec? PA seems to be able to convert from e.g. 32-bit float to aLaw.

Today I use SDR with SpyServer and a local SDR# at the remote RX site. But this implies a PC (SDR# only works on Windows). A RPi as remote RX would be more practical, and we cannot stream IQ samples over costly mobile internet lines.

RTL-airband would be perfect, as it allows monitoring several VHF channels within the RTL-bandwidth, and stream it to a central Point. As a user, I can then select which sites and frequencies to monitor.

rgds
Peter

szpajder, 2019-07-30:
> Yes I know there is Icecast (I used it many yearrs ago, but then only for one channel at a time), but when using multiple RX at the same frequency, Icecast is not really practically usable due to its unpredictable latency.

What's the use case for streaming the same frequency multiple times?

> Is there a negotiation between the Pulseaudio server and RTL-airband to setup the stream parameters?

Nope, there isn't.

> Can RTL-airband send a stream as aLaw or some other format, less bandwidth-consuming than 44kHz/32bitFLOAT?

rtl_airband only produces 8kHz or 16kHz audio (depending on whether NFM support has been enabled at the compilation stage or not). It's the pulseaudio library which resamples the audio to the higher speed.

> PA seems to be able to convert from e.g. 32-bit float to aLaw.

Then it might be possible to do this conversion purely by appropriate pulseaudio configuration (but I haven't tried that).

> RTL-airband would be perfect, as it allows monitoring several VHF channels within the RTL-bandwidth, and stream it to a central Point. As a user, I can then select which sites and frequencies to monitor.

Once the streams are running, the PA server will be playing them all simultaneously. The only selection you can then do is to mute all the streams which you don't want at a particular time. Isn't this cumbersome, if you have something like 20 streams? With Icecast, you just point your player to the stream(s) you want to listen. Done.

With respect to stream latency - indeed, PA can give you low latency streaming, but only if the transport network provides low and stable latency as well. PA does not do a lot of buffering, so even a momentary packet drop or latency increase (which is common on mobile internet links), will cause the audio to get choppy. Icecast is more suitable for this purpose, because the client buffers a couple of seconds of audio, hence if the data stops arriving, audio can still be played continuously before the buffer is emptied. Low latency with jerky audio or continuous audio with more latency - choose your preference, you can't have both.

Also note that:

- there are a few knobs in Icecast server config which you can tune to reduce the delay - see https://github.com/szpajder/RTLSDR-Airband/wiki/Setting-up-your-own-Icecast-server#configure
- most players have configurable input buffer size - that's another knob which affects delay

liveatc.net and broadcastify.com use Icecast and many feeds on these sites are produced with rtl_airband - so maybe Icecast is not that bad ;)

sm5gxq, 2019-07-31:
Hi,

In fact, I'm listening to the same frequency, but from different receivers. A transmitting station is often received on multiple receivers, and if the latency is not roughly the same - then there is the echo effect.

The main reason I wrote you regarding this, was to ask if the streaming parameters in RTL airband could be adjusted. But it seems not, so then I must investigate further into PA. I have not used PA Before. 2 years ago I tried to set up RTL Airband with PA but with no success.

If one uses the Tunnel sink, it would in theory be possible to stream to a remote server. But as far as I can see, the streaming parameters on the tunnel sink also sets the same parameters on the link. And I suppose this would not work, bacause first we need to match  what RTL Airband is sending? Which is fixed to 16 kHz and 32 bit float?

Can you in your configuration file set up different sinks for each frequency?

PA tunnel module works this way, and if I read the PA docs correctly, you can address different sinks at the remote server. But if you are using the RTP plugin, then Everything seems to end up at the same sink :-(.

And of course, there are other streaming alternatives out there, that could get its audio from PA and send it onwards. But we also must consider the "corking"; since we don't want to have the remote RPi stream continously.

rgds
Peter

szpajder, 2019-07-31:
I don't have my notes at hand, but AFAIR during my tests with a single mono stream I have seen a traffic rate of 256 kbps or 512 kbps, depending on NFM being disabled or enabled, respectively. This comes from 8 bits/byte * 4 bytes/float * 8000 samples/sec (or 16000 samples/sec, with NFM enabled). So no, rtl_airband does not stream 44.1 kHz audio. It streams exactly what it produces (8k or 16k, mono or stereo), without compression. No other sample formats are currently supported. If I read PA API doc correctly, PA can accept alaw or ulaw samples directly on input, so sample conversion would need to be done by rtl_airband. I may implement it one day, however 8k or 16k at 8 bit is still plenty of data to transmit, so I still recommend MP3 and Icecast if bandwidth or traffic volume is of a concern.

> Can you in your configuration file set up different sinks for each frequency?

Yes, just use "sink" parameter in the output configuration (https://github.com/szpajder/RTLSDR-Airband/wiki/Configuring-PulseAudio-outputs).

> But we also must consider the "corking"; since we don't want to have the remote RPi stream continously.

It doesn't stream continuously unless "continuous" option in the output config is set to true.

closed 2020-02-24 by szpajder