rtl-airband/RTLSDR-Airband #134: Support no username icecast server

hayden-t, 2019-08-12:
So after some trouble getting this to connect i found the problem was that the stream service im casting to does not require or specify a username, and if one is not set in config you get error, but if you set it to "" i think it is reverting to default shout one, which is "source" 
so to get it to connect i had to comment out the set username code in output:
```
	if (shout_set_user(shouttemp, icecast->username) != SHOUTERR_SUCCESS) {
		shout_free(shouttemp); return;
	}
```
provider is https://www.broadcastify.com  https://www.radioreference.com/

szpajder, 2019-08-25:
Sorry for delay, I was on vacation.

Shout library works as follows:

- initial connection request does not contain any authentication credentials. If the server refuses the connection, the next request will contain the credentials. So if your server does not require authentication at all, you can just configure any credentials - they won't be used anyway.

- if the user name is NULL (not set), the library will use the default one instead, which is "source".

- If the user name is set to an empty string (""), it is sent as is. This value doesn't have any special meaning.

- If the password is NULL, the library will return an error and won't attempt to connect at all.

Therefore commenting out the above code section causes the user name to be reverted to "source". So this modification does not bring any benefit - you can just set `user_name` parameter in the icecast output config to "source" and the result will be the same.

hayden-t, 2019-08-26:
"If the user name is set to an empty string (""), it is sent as is. This value doesn't have any special meaning." I tried this it does not work, i think it still sends, "source"

szpajder, 2019-08-26:
>
> I tried this it does not work, i think it still sends, "source"
>
Can you please verify this by taking a traffic dump? I have taken one and
it clearly shows an empty user name is sent when it's been so configured.

What behavior do you need to make this work? Should no authentication be
done whatsoever or should it authenticate with an empty username and a
non-empty password?

szpajder, 2019-08-28:
> how do i do a traffic dump ?
>
1. Stop rtl_airband
2. sudo tcpdump -As0 -ni name_of_your_network_interface_in_use -c 15 host
ip_address_or_hostname_of_the_icecast_server
3. Start rtl_airband while tcpdump is running

After capturing 15 packets (-c option value) tcpdump will exit. The dump of
the initial 15 packets will be dumped to the screen. Look for the packets
containing HTTP requests and server replies (HTTP is a text-based protocol,
so it's easy to spot). Find the "Authorization: Basic" header, grab its
value (copy the value only, without the header name) and type: "echo
the_auth_value | base64 -di" to decode it and see what username and
password you are actually sending to the server.

> it should authenticate with password only, even username = "" fails auth
> unless i have the lines mentioned commented out...
>
Icecast protocol uses HTTP Basic authentication mechanism. It forms an
authentication string "username:password", encodes it with BASE64 and puts
the resulting string into "Authorization:" header of the HTTP request. So
it's not possible not to send username at all. The best you can do is to
specify an empty username - this will result in a string ":password" to be
sent to the server

You mentioned that you are streaming to Broadcastify, right? I have found
this guide:
https://wiki.radioreference.com/index.php/Raspberry_Pi_RTL-SDR_Broadcastify
- although it it based on an outdated version of rtl_airband, the example
config clearly shows "source" as the username to be used. Is this no longer
valid?

hayden-t, 2019-08-28:
how do i do a traffic dump ?
it should authenticate with password only, even username = "" fails auth unless i have the lines mentioned commented out...

closed 2020-02-24 by szpajder