rtl-airband/RTLSDR-Airband #135: Expose or mention in readme that stream bitrate is set in rtl_airband.h

hayden-t, 2019-08-12:
i see another comment about this, and realize it is not likely a high priority, but you should mention in the readme/wiki where this value is set b4 compile, as my host requires a set bitrate. thanks.

szpajder, 2019-08-25:
Which parameters do you need to modify to make it work?

I can make them configurable, if possible.

hayden-t, 2019-08-26:
you should just mention in the readmen that there are more settings in rtl_airband.h

szpajder, 2019-08-26:
> you should just mention in the readmen that there are more settings in
> rtl_airband.h
>
These variables are not settings. They are not meant to be modified by
users. Whatever is meant to be user-configurable, goes into the config
file. If something is hardcoded, then it's either done for a good reason or
I just couldn't come up with a use case requiring a different setting than
the default (aka. "nobody will ever need to tweak this"). If you do need to
reconfigure something which is not configurable, please explain what and
why and I'll add a knob for it if it's possible and justified. You
mentioned stream bit rate, but in fact this parameter is not set in
rtl_airband.h. It's hardcoded in output.cpp.

szpajder, 2019-08-26:
Nope. This is MP3 sample rate (samples/sec, not bits/sec).

hayden-t, 2019-08-27:
is this not the stream bitrate ? https://github.com/szpajder/RTLSDR-Airband/blob/master/rtl_airband.h#L68

closed 2020-02-24 by szpajder