rtl-airband/RTLSDR-Airband #136: Unable to get clean audio

dh47, 2019-09-17:
I have been trying so hard to be able to record and stream a clean audio signal with no luck what so ever and it's driving me COMPLETELY NUTS!

No matter what I do I either get buzzing or some humming sound and distorted audio.

Where is my config.
`tau = 100;
devices:
({
  type = "rtlsdr";
  index = 0;
  gain = 16;
 #mode = "scan";
  sample_rate = 2.4;
  centerfreq = 141.400;
  channels:
  (
    {
      freq = (141.400);
      labels = ("EAS");
	  modulation = "nfm";
      outputs: (
        {
	  type = "icecast";
	  server = "icecast.server.example.org";
          port = 8080;
          mountpoint = "stream.mp3";
          name = "EAS";
          genre = "EAS";
          description = "EAS";
          username = "source";
          password = "mypassword";
		  send_scan_freq_tags = false;
        },
        {
          type = "file";
          directory = "/home/pi/radio";
          filename_template = "EAS";
        }
      );
    }
  );
 }
);`

What the heck am I doing wrong?

szpajder, 2019-09-17:
First of all:

- don't put parentheses around freq value. It's not a list, its a single value, so it should be just `freq = 141.000;`
- don't set `centerfreq` to the same value as `freq`. Shift it at last a couple tenths of kilohertz either side (for example 141.450 would be a good value).
- no `correction` value has been specified. Are you sure your dongle does not need frequency correction?
- `gain = 16` might be a little low, unless your signal of interest is very strong or you are using an external amplifier or a directional antenna.
- `labels` parameter does nothing unless `mode` is set to `scan`.

dh47, 2019-09-17:
Freq value changed, centerfreq changed.

No correction from what I have been testing in SDR sharp.
gain = 16 yep correct, the repeater is only a few miles away from me.

spotter-ssol, 2019-09-17:
Can you post an audio sample?

dh47, 2019-09-17:
Only a small one but you can hear the hum.
[EAS_20190917_17.zip](https://github.com/szpajder/RTLSDR-Airband/files/3622801/EAS_20190917_17.zip)

szpajder, 2019-09-17:
Nothing wrong here, I guess it's https://en.wikipedia.org/wiki/CTCSS ?

dh47, 2019-09-17:
You cant hear that hum?

szpajder, 2019-09-17:
I can hear it, and I have given you a link where you can read what it is.

dh47, 2019-09-17:
Yeah, it does use CTCSS and the value of it is 186.2.  Would be cool to remove that noise.

dh47, 2019-09-17:
Resolved after looking into CTCSS here. By adding the following to my config.
lowpass = 2500;
highpass = 150;
squelch = 50;

closed 2019-09-18 by dh47