rtl-airband/RTLSDR-Airband #137: rtl_airband service fails to start on boot

Charlier26, 2019-10-09:
I have been unable to start rtl_airband as a service using a SDRplay RSP1A with SoapySDR. I can start the service manually, and it runs fine, but not automatically at boot. Here is what I get back from journalctl : 

-- Logs begin at Thu 2019-02-14 10:11:59 GMT, end at Wed 2019-10-09 19:05:03 BST. --
Oct 09 19:03:18 SDRPlay systemd[1]: Started SDR AM/NFM demodulator.
Oct 09 19:03:20 SDRPlay rtl_airband[388]: libusb: error [udev_hotplug_event] ignoring udev action bind
Oct 09 19:03:21 SDRPlay rtl_airband[388]: SoapySDR: device 'driver=sdrplay, hardware=18070CA796, rfgain_sel=6': using native sample format 'CS16' (fullScale=32767.0)
Oct 09 19:03:21 SDRPlay rtl_airband[388]: RTLSDR-Airband version v3.0.1-20-ga93dc1d starting
Oct 09 19:03:22 SDRPlay rtl_airband[388]: [INFO] Using format CS16.
Oct 09 19:03:22 SDRPlay rtl_airband[388]: pulse: 127.0.0.1: connecting...
Oct 09 19:03:22 SDRPlay rtl_airband[388]: SoapySDR: device 'driver=sdrplay, hardware=18070CA796, rfgain_sel=6': sample rate set to 3000000 sps
Oct 09 19:03:22 SDRPlay rtl_airband[388]: SoapySDR: device 'driver=sdrplay, hardware=18070CA796, rfgain_sel=6': AGC off (requested: off)
Oct 09 19:03:22 SDRPlay rtl_airband[388]: SoapySDR: device 'driver=sdrplay, hardware=18070CA796, rfgain_sel=6': gain set to 35.0 dB
Oct 09 19:03:22 SDRPlay rtl_airband[388]: SoapySDR: device 'driver=sdrplay, hardware=18070CA796, rfgain_sel=6' initialized
Oct 09 19:03:22 SDRPlay rtl_airband[388]: SoapySDR: device 'driver=sdrplay, hardware=18070CA796, rfgain_sel=6' started
Oct 09 19:03:22 SDRPlay rtl_airband[388]: pulse: 127.0.0.1: connection failed: Connection refused
Oct 09 19:03:32 SDRPlay rtl_airband[388]: pulse: 127.0.0.1: connecting...
Oct 09 19:03:32 SDRPlay rtl_airband[388]: pulse: 127.0.0.1: connection failed: Connection refused
Oct 09 19:03:34 SDRPlay rtl_airband[388]: Cannot open output file /home/pi/recordings/TWR+GND+APP_20191009_18.mp3 (No such file or directory), output disabled
Oct 09 19:03:42 SDRPlay rtl_airband[388]: pulse: 127.0.0.1: connecting...
Oct 09 19:03:42 SDRPlay rtl_airband[388]: pulse: 127.0.0.1: connection failed: Access denied
Oct 09 19:03:57 SDRPlay rtl_airband[388]: SoapySDR device 'driver=sdrplay, hardware=18070CA796, rfgain_sel=6': readStream failed: TIMEOUT, disabling
Oct 09 19:03:57 SDRPlay rtl_airband[388]: All receivers failed, exiting
Oct 09 19:03:57 SDRPlay rtl_airband[388]: Freeing GPU memory
Oct 09 19:03:57 SDRPlay rtl_airband[388]: Cleaning up
Oct 09 19:03:57 SDRPlay rtl_airband[388]: Input threads closed
Oct 09 19:03:57 SDRPlay systemd[1]: rtl_airband.service: Succeeded.

Any suggestions?

szpajder, 2019-10-11:
It started correctly and was running for about 30 seconds, then it failed to read samples from the device, so it declared the device as failed. As there were no other devices in operation, it exited. The timeout value is 1 second, so it's quite a lot - if the device failed to provide any samples for that long, maybe it got disconnected or is working intermittently? This might be an indication of power problems. Check your USB cabling, power cabling, try another power supply.

> I can start the service manually, and it runs fine, but not automatically at boot

During boot the CPU is probably busy starting various services, so the power consumption might be higher than later, when the boot is complete. If you run a lot of tasks on this Pi, you can also try disabling some of them to reduce the load.

Charlier26, 2019-10-11:
Thanks for the reply. Can you suggest a better way of starting it, perhaps after the Desktop loads? I’m kind of a noob with Raspbian. I tried making a shell script to be called by rc.local, which delayed running /home/pi/RTLSDR-Airband/rtl_airband until after a sleep of 30, but it didn’t run.

 

Charlie

ScanCaster, 2019-10-12:
>I can start the service manually, and it runs fine, but not automatically at boot

Not a fan of systemd, so I don't use it to auto start so I just do it old school. cron.

You can create a BASH script and have it called on reboots...

in cron use the @reboot to replace the DOMT settings and just spec your shell script to call up airband

@reboot /home/pi/startAirband.sh
*/3 * * * * /home/pi/CheckAirband.sh

Example startAirband.sh
#!/bin/bash

/usr/local/bin/rtl_airband -c MyAriband.confg

Just to confirm if you are using systemd files you USED /usr/local/bin/rtl_airband -F -c MyAIrband.conf

The -F option is needed for service use ie: starting via systemd.

Is there a reason you start X??? If not goto raspi-config and turn it off to just boot to runlevel 3

rc.local should work similar to cron as well..

What is your shell script?

I use @reboot to start on reboots, and then a */3 in cron to verify its running for any crashes that might happen etc.. I do this on a couple things like with darkice, rtl_fm/multimon setup... stuff I need running 24/7/365.

Charlier26, 2019-10-12:
This one I use with both murmur (which starts as a service) and Mumble (which I have YET to figure out how to autostart, so that I can log into the murmur server from my phone/laptop, etc to listen. I’m not sending to LiveATC with this one.

 

Here is the shell script I tried:

 

#!/bin/sh

 

Sleep 15

sudo systemctl start /home/pi/RTLSDR-Airband/rtl_airband

 

 

I run X because I am new to linux and am used to using VNC to my windows machines to look at things. I use SSH with the Pi, but haven’t had any luck running things via SSH then putting them into background using CTL-Z then bg. I also need X to start Mumble on the Pi because I haven’t found a way to autostart it.

 

Charlie

Charlier26, 2019-10-12:
Whoops.....I sent the wrong script......THIS is the one I was trying. First it has to RESTART mumble-server because until you restart it, it refuses all connections (don't know why). Then, starting rtl_airband, and then mumble

#!/bin/sh -e
sleep 30

sudo systemctl restart mumble-server &

sleep 30

sudo /home/pi/RTLSDR-Airband/rtl_airband &

sleep 15

mumble mumble://pi:xxx@127.0.0.1:64738/LAN/Root/N &

Charlie

Charlier26, 2019-10-12:
Also, just to make sure I'm not messing things up in the .conf file, here it is:

cat /usr/local/etc/rtl_airband.conf
# Scanning mode example
# Single dongle, three frequencies, output to Icecast server and to a file.
# Refer to https://github.com/szpajder/RTLSDR-Airband/wiki
# for description of keywords and config syntax.

devices:
({
  type = "soapysdr";
  device_string = "driver=sdrplay, hardware=18070CA796, rfgain_sel=6";
  index = 0;
  gain = 35;
  correction = 80;
  mode = "scan";
  channels:
  (
    {
      freqs = ( 118.7, 119.7, 121.0, 124.9, 128.7 );
      squelch = ( 42,-1, -1, -1, -1 );
      labels = ( "Tower", "Ground", "Approach", "Approach2", "Approach3" );
      outputs: (
        {
	  type = "pulse";
          server = "192.168.1.134";
	  sink = "alsa_output.platform-soc_audio.analog-mono";
        }
      );
    }
  );
 }
);

pi@SDRPlay:~ $

ScanCaster, 2019-10-12:
I think your script has a problem in the PATH for rtl_airband

SSH in... and then do

which rtl_airband

You should get:

/usr/local/bin/rtl_airband

Unless you specified to build it some place else when you compiled it? Did you follow the compiling instructions here? Or something else????

So the reason it doesn't start it you are using

sudo /home/pi/RTLSDR-Airband/rtl_airband

First, you don't need SUDO... you need to do THIS AS ROOT..

After you ssh in 

sudo su
enter your sudo password which is the same as your normal one...

Second, the path /home/pi/RTLSDR-Airband

do an ls -larth /home/pi/RTLSDR-Airband 

is there a program rtl_airband in there? AND did the which command above give you that SAME path?

CHANGE your script to:
/usr/local/bin/rtl_airband -c /home/pi/otherdir/yourconfigfile.conf

OR 
/path/which/gave/you/rtl_airband -c /home/pi/otherdir/yourconfigfile.conf

I don't think you need the & to send to background when I run it it goes to background without the -f option...

If that works, make the APPROPRIATE changes in to the path in your systemd files etc.. 

I don't know what mumble is, so I am not touching that...nor the config file for rtl_airband...

Charlier26, 2019-10-12:
Yes, the program is in /usr/local/bin. But yes, there is another one in /home/pi/RTLSDR-Airband.
Turns out that the problem is with pulseaudio. I can get rtl_airband to start automatically, but when I look at the log, pulseaudio denys access to it unless I manually start it from a terminal window. Been reading up on pulseaudio to try and figure out why it denies access. (I think it might have something to do with the cookie)

Charlie

szpajder, 2019-10-13:
@Charlier26 you can also get automatic restarting feature via systemd - just modify the `rtl_airband.service` file (it's probably in `/etc/systemd/system` directory), change `Restart=no` to `Restart=always`.

closed 2020-02-24 by szpajder