rtl-airband/RTLSDR-Airband #140: Squelch+delay

alex512125, 2019-11-10:
Hi!
I have just tried your project and it is very good but can you add or say me how i can change something.
1) I want to add some delay after squelch turns on.For example 1 second after speech breaks.
2)I want to make squelch harder. I mean that now it can cut some speech at the end and changing to manual squelch doest help (in sdr# you changing squelch number change squelch more smoothly ).
Thanks for any answer!

szpajder, 2019-11-10:
Currently there is auto squelch and manual squelch and that's it. There are no more configuration knobs for squelch as of today, sorry.

alex512125, 2019-11-10:
I can't be problem in squelch. For example there is speech.Squelch is off. Then squelch turns on immediately.But i want set 0.1-0.4 seconds squelch off after speech breaks and add 0.6 second pause.

LoganSound, 2021-01-07:
Interested in this as well - currently looking at trashing files that fall under a certain length threshold but "squelch release" would offer more control.

SullivanChrisJ, 2025-07-21:
If someone is ever thinking about adjustable squelch delay and/or hang time I'd like to see negative values, to enable pre-roll and post-roll. I am interesting in transmitter fingerprinting through analysis of the raw files and what happens at the very beginning of transmission can be very enlightening.