rtl-airband/RTLSDR-Airband #141: error when make PLATFORM=rpiv2 WITH_SOAPYSDR=1

enghammer, 2020-01-31:
I have a RPi3 runing with RTL-SDR. and RPi4 with AirSpy..
Now I want to move the AirSpy to the RPi3 and at the end, I get this tree errors when compiling:  

input-soapysdr.cpp:315:37: error: invalid conversion from ‘SoapySDRStream**’ to ‘int’ [-fpermissive]
  if(SoapySDRDevice_setupStream(sdr, &rxStream, SOAPY_SDR_RX, dev_data->sample_format,
                                     ^~~~~~~~~
input-soapysdr.cpp:316:45: error: invalid conversion from ‘int’ to ‘const char*’ [-fpermissive]
  (size_t * const)&dev_data->channel, 1, NULL) != 0) {
                                             ^
input-soapysdr.cpp:315:72: error: cannot convert ‘const char*’ to ‘const size_t*’ {aka ‘const unsigned int*’}
  if(SoapySDRDevice_setupStream(sdr, &rxStream, SOAPY_SDR_RX, dev_data->sample_format,
                                                              ~~~~~~~~~~^~~~~~~~~~~~~
In file included from input-soapysdr.cpp:29:
/usr/local/include/SoapySDR/Device.h:298:19: note:   initializing argument 4 of ‘SoapySDRStream* SoapySDRDevice_setupStream(SoapySDRDevice*, int, const char*, const size_t*, size_t, const SoapySDRKwargs*)’
     const size_t *channels,
     ~~~~~~~~~~~~~~^~~~~~~~
make: *** [<builtin>: input-soapysdr.o] Error 1

Any Ideas ??
Thanks..

szpajder, 2020-02-01:
rtl_airband was using old `SoapySDRDevice_setupStream()` syntax which has recently been removed from SoapySDR code. Thanks for pointing this out.

Please use rtl_airband unstable branch, I've just pushed a fix there.

enghammer, 2020-02-01:
Perfect, That did it...
Thanks a lot...

closed 2020-02-01 by enghammer