rtl-airband/RTLSDR-Airband #142: Feed clipping

chrisjob1021, 2020-02-10:
Hi, I am running a feed on latest version 3.1.0 and I am encountering feed clipping. This issue predates the latest version, but I was optimistic it would help relating to squelch changes. 

I'm in multichannel scan mode. There should be no issues with the receive strength, because you can clearly hear the transmission once the clipping ends. 

I've tried tuning gain (up and down) and fft_size (up and down) to no avail. I'm using -1 for squelch. I've tried setting various squelch settings manually, but I find that some days it will work and other days it results in channel staying open for too long. 

Have a listen about 8:40 in with SW2414 on the missed approach: https://archive-server.liveatc.net/kbur/KBUR3-Gnd-Twr-App-Feb-10-2020-0730Z.mp3

This happens on all channels, but especially on the ones I receive really well. 

I'm using following hardware:
Raspberry PI 3 Model B 1.2GHz 64-bit quad-core ARMv8 CPU, 1GB RAM
NooElec NESDR Mini 2 SDR & DVB-T USB Stick (RTL2832 + R820T2)

hamav8tor, 2020-02-10:
I don't hear or see any clipping in this recording. Clipping would result in distortion and is usually only experienced when the gain is set too high on an SDR and signals are unusually strong (like overheard or very close on the ground).

tinder-chrisobrien, 2020-02-10:
Try listening a bit longer. Maybe 30-60 seconds. 

I’ve tried playing with the gain higher or lower. It does affect the transmissions I receive the best (e.g. Burbank tower, ground, SoCal controller).

chrisjob1021, 2020-02-10:
Try listening a bit longer. Maybe 30-60 seconds.

I’ve tried playing with the gain higher or lower. It does affect the transmissions I receive the best (e.g. Burbank tower, ground, SoCal controller).

chrisjob1021, 2020-02-10:
I messed up. It’s 9:40-10:40 into that clip.

hamav8tor, 2020-02-10:
I checked the whole recording. Still not hearing what you are hearing. In that segment I hear a controller and then a somewhat weaker signal aircraft response. The aircraft transmission briefly falls out of squelch and then back when the signal gets stronger.

tinder-chrisobrien, 2020-02-10:
@hamav8tor This also happens with stronger ATC communication (e.g. controllers). It seems to fall out of squelch, when it really shouldn't. I'll find another example and pass along in case you find helpful.

chrisjob1021, 2020-02-10:
@hamav8tor This also happens with stronger ATC communication (e.g. controllers). It seems to fall out of squelch, when it really shouldn't. The controllers are loud and clear. I'll find another example and pass along in case you find helpful.

szpajder, 2020-02-11:
The demodulator and scanning controller in this program run completely independently - neither of them knows what the other one is doing at the moment. The demodulator does not know precisely, which scanned channel the given sample batch came from. This is an inherent limitation of batching FFTs into large blocks to be computed on the GPU. So you may have an impression that the scanner has resumed scanning despite strong transmission on the current channel, but in fact it might have resumed due to no transmission being active on the previous channel in sequence. Similar issues have been reported before (like https://github.com/szpajder/RTLSDR-Airband/issues/88 or https://github.com/szpajder/RTLSDR-Airband/issues/138).

To put it short: every piece of hardware should be used for the purpose it is good at. Superheterodyne scanners - for scanning, wideband SDRs with low power computers - for multichannel mode.

chrisjob1021, 2020-02-11:
@szpajder Thanks for the feedback. It's very helpful. I will try as suggested in #138.

closed 2020-02-11 by chrisjob1021