rtl-airband/RTLSDR-Airband #144: Problem with v3.1 on Raspbian Buster

mglimmer, 2020-02-27:
I have tried building the new 3.1 version on a  RPi 4 with Raspbian Buster using the same method I used with v3.01 and I am having a problem.  The build appears to be successful, but when testing it using rtl_airband -f, it returns to a command prompt instead of going to the waterfall display.

The output before returning to the command prompt is:
    Detached kernel driver
    Found Rafael Micro R820T tuner
    [R82XX] PLL not locked!
    Allocating 10 zero-copy buffers

Any suggestions what I am doing wrong?

szpajder, 2020-02-28:
Run the program with -F -e instead of -f to get the logs printed to the
console. Most probably you will see something like:

Unable to enable V3D. Please check your firmware is up to date.

This is because the GPU on the RPiv4 is quite different to the one found on
the older RPi generations and the accelerated FFT code in rtl_airband does
not work on it. Please use the non-accelerated version (ie. make clean and
then compile again with PLATFORM=armv7-generic).

mglimmer, 2020-02-28:
Thank you! Moving the FFT from the GPU to the CPU resolved all of the problems.

I used the option PLATFORM=armv8-generic, because the RPi4 should support 64-bit.

If you update the WIKI to note the RPi4 changes, you may also want to note the library dependency that is needed in Raspbian to run the FFT code on the CPU:
sudo apt-get install libfftw3-dev

szpajder, 2020-02-29:
Wiki has been updated.

Closing.

closed 2020-02-29 by szpajder