rtl-airband/RTLSDR-Airband #146: Building on macOS?

ajmas, 2020-03-09:
I am trying to build the project on macOS 10.15.3, along with MacPorts. So far I have the following:

```
sudo port install fftw-3 
sudo port install libconfig-hr
sudo port install libshout2

export CPLUS_INCLUDE_PATH=/opt/local/include:/usr/include
PLATFORM=x86 make 
```

This results in the following failure:

```
c++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.1.0\" -march=native -DWITH_RTLSDR   -c -o util.o util.cpp
util.cpp:142:3: error: use of undeclared identifier 'sincosf'; did you mean '__sincosf'?
                sincosf(2.0f * M_PI * (float)i / 256.0f, sin_lut + i, cos_lut + i);
                ^~~~~~~
                __sincosf
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:650:29: note: '__sincosf' declared here
__header_always_inline void __sincosf(float __x, float *__sinp, float *__cosp) {
                            ^
1 error generated.
make: *** [util.o] Error 1
```

Any suggestions for resolving this would be appreciated.

ajmas, 2020-03-09:
Have added the following to `util.cpp` which solved that issue:

```
#if __APPLE__
  #define sincosf __sincosf
#endif
```

Now need to deal with a new set of errors, which I'll see if I can fix:

```
c++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.1.0-dirty\" -march=native -DWITH_RTLSDR   -c -o util.o util.cpp
In file included from util.cpp:29:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:318:9: error: no member named 'signbit' in the global namespace
using ::signbit;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:319:9: error: no member named 'fpclassify' in the global namespace
using ::fpclassify;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:320:9: error: no member named 'isfinite' in the global namespace; did you mean 'finite'?
using ::isfinite;
      ~~^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:749:12: note: 'finite' declared here
extern int finite(double)
           ^
In file included from util.cpp:29:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:321:9: error: no member named 'isinf' in the global namespace
using ::isinf;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:322:9: error: no member named 'isnan' in the global namespace
using ::isnan;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:323:9: error: no member named 'isnormal' in the global namespace
using ::isnormal;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:324:7: error: no member named 'isgreater' in the global namespace; did you mean '::std::greater'?
using ::isgreater;
      ^~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/functional:728:29: note: '::std::greater' declared here
struct _LIBCPP_TEMPLATE_VIS greater : binary_function<_Tp, _Tp, bool>
                            ^
In file included from util.cpp:29:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:325:7: error: no member named 'isgreaterequal' in the global namespace; did you mean
      '::std::greater_equal'?
using ::isgreaterequal;
      ^~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/functional:757:29: note: '::std::greater_equal' declared here
struct _LIBCPP_TEMPLATE_VIS greater_equal : binary_function<_Tp, _Tp, bool>
                            ^
In file included from util.cpp:29:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:326:9: error: no member named 'isless' in the global namespace
using ::isless;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:327:9: error: no member named 'islessequal' in the global namespace
using ::islessequal;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:328:9: error: no member named 'islessgreater' in the global namespace
using ::islessgreater;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:329:9: error: no member named 'isunordered' in the global namespace
using ::isunordered;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:330:9: error: no member named 'isunordered' in the global namespace
using ::isunordered;
      ~~^
13 errors generated.
make: *** [util.o] Error 1
```

beaugunderson, 2020-03-26:
@ajmas were you able to solve this? i'm about to start down the same path :)

ajmas, 2020-03-26:
Not yet. I have not looked at it in a couple of weeks. 

Do add anything you learn here.

ah1102, 2020-06-22:
I join the topic

ajmas, 2020-06-22:
@ah1102 what do you mean by your comment, given macOS is fully running on Intel x86 processors?

ah1102, 2020-06-22:
Don’t you know that Apple has refused to support 32-bit applications since Mojava or earlier? The operating system will simply not let you run any 32 bit code. For this reason, for example, Wine Skin and other old programs have stopped working.

> @ah1102 what do you mean by your comment, given macOS is fully running on Intel x86 processors (currently i5, i7 and Xeon)?

ajmas, 2020-06-22:
@ah1102 not supporting 32-bit code is indeed true, but saying not supporting x86 doesn’t say the same thing. 

For many developers x86 indicates the instruction set and not the bitness of the addressing. 

Given the Rasberry Pi is also moving towards 64-bit (albeit on ARM), maybe we should open a ticket for this specifically?

charlie-foxtrot, 2020-06-22:
@ah1102 I'm not sure I follow.

Are you saying you cant compile this source on a Mac because the source code itself is 32-bit?

I understand that you cant run a 32-bit binary, but if you compile source on your Mac with the compiler installed from Xcode command line tools, that would generate a 64-bit binary, right?

ajmas, 2020-06-22:
@charlie-foxtrot so it would seem there has been some support building [64-bit on Raspbian](https://github.com/szpajder/RTLSDR-Airband/issues/166), so I think 64-bit may be a red herring.

I think the best way to approach this is to make a reference build on Raspbian, and then maybe use Homebrew instead of MacPorts and then compare notes? I just need to retool (I am currently MacPorts based).

ah1102, 2020-06-22:
I do not know the details. It just seemed strange to me that you are trying to make x86 on macOS Catalina (x64 only)

szpajder, 2021-10-17:
The unstable branch of rtl_airband should now compile on MacOS 10.15 and 11 (at least https://github.com/szpajder/RTLSDR-Airband/actions/runs/1352182661 says so).

Note that the build procedure has changed in unstable. The program is now built with cmake. Here is a quick HOWTO before the wiki gets updated:

```text
brew install lame libshout libconfig fftw librtlsdr soapysdr pulseaudio     # last three are optional
git clone https://github.com/szpajder/RTLSDR-Airband.git
cd RTLSDR-Airband
mkdir build
cd build
cmake ../
make -j
make install
```