rtl-airband/RTLSDR-Airband #149: Wiki documentation on lowpass and highpass are flipped

charlie-foxtrot, 2020-04-15:
Wiki documentation for `lowpass` and `highpass` are flipped.  `lowpass` should be a higher value than `highpass`, ie:

```
channels: (
  {
    freq = 123.00;
    lowpass = 3200;
    highpass = 200;
    outputs: ( ... );
  },
  ...
);
```

FWIW the default values in the code are correct: https://github.com/szpajder/RTLSDR-Airband/blob/5fdf59e2f2ec4cebbce917b91fc92afb149668c9/config.cpp#L210-L211

szpajder, 2020-04-16:
Where?

Here?

https://github.com/szpajder/RTLSDR-Airband/wiki/Audio-filters-in-MP3-outputs

Nope, it's correct.

charlie-foxtrot, 2020-04-16:
Yes on that page, in the code block.

`lowpass` passes frequencies lower than the value.  That link shows `lowpass` being set to `200` instead of being set to `3200`.

Similarly it shows `highpass` being set to `3200` instead of `200`.  If you use the code snippet from that page you will end up with nothing but silence (coincidently how I discovered the it).

szpajder, 2020-04-16:
OK, I now realized that you've pasted what should be there, not what is there now and this of course didn't match what I was seeing there :)

Thanks, fixed.

closed 2020-04-16 by szpajder