rtl-airband/RTLSDR-Airband #153: Wiki documentation missing for 'Configuring Icecast outputs'

charlie-foxtrot, 2020-04-30:
Icecast output supports a `send_scan_freq_tags` but not mentioned in the Wiki section

szpajder, 2020-05-01:
https://github.com/szpajder/RTLSDR-Airband/wiki/Icecast-metadata-updates-in-scan-mode

charlie-foxtrot, 2020-05-01:
same for this one, I missed it.  maybe add a link from https://github.com/szpajder/RTLSDR-Airband/wiki/Configuring-Icecast-outputs

the "Explanation of keywords" section missing one of the keywords threw me off

szpajder, 2020-05-03:
If you missed it, then how did you know there is such a feature in the first place? :)

I avoid creating a full mesh of cross-links between pages in the Wiki because it disrupts the reading order and confuses beginners. There is a complete table of contents on the first page, which also shows up in the sidebar and that's enough.

charlie-foxtrot, 2020-05-03:
> If you missed it, then how did you know there is such a feature in the first place? :)

noticed in the code then went back to the wiki looking for a description.

szpajder, 2020-08-04:
As I said. I prefer to keep the main page of each topic short and simple, so that it does not act as a visual deterrent for newcomers. All the non-essential features have their respective pages in the "Advanced features" section.

closed 2020-08-04 by szpajder