rtl-airband/RTLSDR-Airband #155: update icecast metadata for Mixers

charlie-foxtrot, 2020-05-01:
Icecast metadata updates happen in scan mode but not for Mixers.  It would be nice if an icecast stream out of a mixer included the channel(s) that are currently broadcasting.

charlie-foxtrot, 2020-05-01:
I'm working on a PR to add `label` to the "mixer" output for a channel then in the mixer thread send metadata updates.  Following the `send_scan_freq_tags` model this will be enabled behind a flag.

Questions:
1. Should I add a second flag (like `send_mixer_freq_tags`) or rename the existing tag (like `send_freq_tags` and have it apply for both scan and mixer?
2. From the mixer thread is there a way to know if an input is "active" or not?  It seems through each loop of `mixer_thread()` each input always has something even if there is no activity on the channel and `channel->axcindicate` is always set to `'*'`.  What I have found to work is adding a new parameter to `mixer_put_samples()` if the input is active or not.

szpajder, 2020-05-03:
1. I prefer `send_freq_tags` for both scan and mixer, however `send_scan_freq_tags` should be kept around for some time as a backward compatibility option.

2. There is now way to know if the mixer input is active or not, because this information was not needed there so far.

Apart from technical details - is there a use case for having this feature for mixers? I didn't implement it, because I couldn't find one. If the information is meant to be digested by the user in real time, then the amount of it shouldn't be overwhelming (in fact just listening to the mixed audio can be overwhelming at times).

charlie-foxtrot, 2020-05-03:
> Apart from technical details - is there a use case for having this feature for mixers? I didn't implement it, because I couldn't find one. If the information is meant to be digested by the user in [...]

I have a pi4 with 3 dongles picking up a total of 24 channels . . . . they are idle most of the time, but yeah, running them all through a single mixer is a bit overwhelming. . . . my hope was actually seeing what frequency is actually broadcasting.

charlie-foxtrot, 2020-05-03:
also, any objection to c++ classes and/or std::string?

szpajder, 2020-05-04:
Let's be honest, this program is not written in C++ :-)

charlie-foxtrot, 2020-08-03:
Instead of going this route I ended up using #168 to watch current activity while streaming

closed 2020-08-03 by charlie-foxtrot