rtl-airband/RTLSDR-Airband #156: Compile Error

M0CRT, 2020-05-05:
Hi

Attempting to compile on the back of Ubuntu 19.04 and am seeing the following error.  Only seems to be an issue with adding the Soapy option.

I'm ensuring the pre-reqs are complete and I'm using the 2.13 SDRplay API Library.

Any thoughts?

Thanks

g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=native -DNFM -DPULSE -DWITH_RTLSDR -DWITH_SOAPYSDR   -c -o input-soapysdr.o input-soapysdr.cpp
input-soapysdr.cpp: In function 'void* soapysdr_rx_thread(void*)':
input-soapysdr.cpp:315:37: error: invalid conversion from 'SoapySDRStream**' to 'int' [-fpermissive]
  if(SoapySDRDevice_setupStream(sdr, &rxStream, SOAPY_SDR_RX, dev_data->sample_format,
                                     ^~~~~~~~~
input-soapysdr.cpp:316:45: error: invalid conversion from 'int' to 'const char*' [-fpermissive]
  (size_t * const)&dev_data->channel, 1, NULL) != 0) {
                                             ^
input-soapysdr.cpp:316:45: error: cannot convert 'const char*' to 'const size_t* {aka const long unsigned int*}' for argument '4' to 'SoapySDRStream* SoapySDRDevice_setupStream(SoapySDRDevice*, int, const char*, const size_t*, size_t, const SoapySDRKwargs*)'
<builtin>: recipe for target 'input-soapysdr.o' failed
make: *** [input-soapysdr.o] Error 1
The command '/bin/sh -c cd / &&   wget https://github.com/szpajder/RTLSDR-Airband/archive/v3.1.0.tar.gz &&   tar -xzf v3.1.0.tar.gz &&   cd /RTLSDR-Airband-3.1.0 &&   make PLATFORM=x86 PULSE=1 NFM=1 WITH_SOAPYSDR=1 &&   make install &&   cd /' returned a non-zero code: 2

starvald, 2020-05-05:
Hi, have you tried building from the unstable branch?

szpajder, 2020-05-05:
Duplicate of https://github.com/szpajder/RTLSDR-Airband/issues/141

closed 2020-05-05 by szpajder