rtl-airband/RTLSDR-Airband #157: unsupported device type

degupukas, 2020-05-23:
:~/RTLSDR-Airband/RTLSDR-Airband$ sudo /usr/local/bin/rtl_airband -f
Configuration error: devices.[0]: unsupported device type

szpajder, 2020-05-24:
My crystal ball is temporarily out of service.

Please post your config file (without passwords, URLs, etc).

szpajder, 2020-08-04:
Closing due to lack of activity.

closed 2020-08-04 by szpajder

rapbando, 2020-08-13:
I'm having the same problem! 

```
radar@radar:~/RTLSDR-Airband-unstable$ sudo rtl_airband -f
Configuration error: devices.[0]: unsupported device type

```
rtl_airband.conf here 

```

 This is a minimalistic configuration file for RTLSDR-Airband.
# Just a single RTL dongle with two AM channels in multichannel mode.
# Each channel is sent to a single Icecast output.
# Refer to https://github.com/szpajder/RTLSDR-Airband/wiki
# for description of keywords and config syntax.

devices:
({
  type = "soapysdr";
  device_string = "driver=hackrf,device_id=0,serial=000000000000000075b068dc32331707";
# channel = 0;
  gain = 25.0;
# gain = "LNA=15,MIX=15,VGA=5";
# correction = 0.0;
# antenna = "";
# mode = "multichannel";
# sample_rate = 2.56;
  centerfreq = 120.0;
  channels:

   (
    {
      freq = 119.5;
      outputs: (
        {
          type = "icecast";
          server = "icecast.server.example.org";
          port = 8080;
          mountpoint = "TWR.mp3";
          name = "Tower";
          genre = "ATC";
          username = "source";
          password = "mypassword";
        }
      );
    },
    { 
      freq = 120.225;
      outputs: (
        {
          type = "icecast";
          server = "icecast.server.example.org";
          port = 8080;
          mountpoint = "GND.mp3";
          name = "Ground";
          genre = "ATC";
          description = "My local airport - ground feed";
          username = "source";
          password = "mypassword";
        }
      );
    }
  );
 }
);
```

szpajder, 2020-08-14:
@rapbando recompile the program with SoapySDR support enabled.

https://github.com/szpajder/RTLSDR-Airband/wiki/Compilation#sdr-hardware-support

rapbando, 2020-08-14:
> @rapbando recompile the program with SoapySDR support enabled.
> 
> https://github.com/szpajder/RTLSDR-Airband/wiki/Compilation#sdr-hardware-support

I already did that, that's why I was getting [this](https://github.com/szpajder/RTLSDR-Airband/issues/181#issue-678741350) error, so I switched to unstable branch.

rapbando, 2020-08-14:
But something is broken because even with basic config for RTLSDR I'm getting the same unsupported device type error.

szpajder, 2020-08-14:
Your config file is fine. This error can only show up in one scenario - when the value of the "type" parameter is unknown. "soapysdr" is a perfectly valid type. If it's unknown, it means the program was not built with WITH_SOAPYSDR=1 option enabled. Or maybe it has been, but you are launching an old binary, which does not have it built in. So make clean and do it once again, correctly this time.

rapbando, 2020-08-14:
Working!!!