rtl-airband/RTLSDR-Airband #158: Raspberry PI4

blackwellj, 2020-05-28:
It doesnt appear to work on a PI 4, Is this in the pipeline?

danmilon, 2020-05-28:
it's impossible to provide assistance without any info other than "it doesn't work". 
which build target did you use? what is the output when running it? any error messages?

blackwellj, 2020-05-28:
Appologies, I have tried to build for a PI4 by using the RPIv2 tag. It builds correctly but does not run. I dont have the PI 4 powered up any more as I have gone back to the PI3b+ which works fine

szpajder, 2020-05-28:
Duplicate of https://github.com/szpajder/RTLSDR-Airband/issues/144

See also https://github.com/szpajder/RTLSDR-Airband/wiki/Compilation#compile-the-program

closed 2020-05-28 by szpajder

rec9140, 2020-05-28:
If you look at one of the issues tickets this came up before.

Pi4 changed the GPU and the FFT library used to use the GPU for FFT has not been updated to the Pi4's GPU as the libs/code/blob needed have not been released and/or added?

Compile it using the Pi2 or Pi using NON GPU mode.