rtl-airband/RTLSDR-Airband #160: Subtle change in project description

Strykar, 2020-06-12:
1. Change
`RTLSDR AM demodulator, support multiple channels per dongle, intended for x86 and Raspberry Pi`
to
`RTLSDR AM demodulator, support multiple channels per dongle, intended for x86 and ARM`

I've been running it on an [Odroid-C2](https://www.hardkernel.com/shop/odroid-c2/) (ARMv8) with no issues for a long time.
Perhaps it would be more appropriate to change `Raspberry Pi` to `ARM` and add Odroid as keyword?.

2. Does GPU FFT work with the newest RPi4?

danmilon, 2020-06-12:
About RPi4, no. See #144

szpajder, 2020-06-12:
@Strykar thanks for pointing this out. I changed the description.

As mentioned by @danmilon, GPU FFT does not support RPiv4's GPU, however you can still run the program using just the main CPU.

closed 2020-06-12 by szpajder