rtl-airband/RTLSDR-Airband #164: Problem with installation on Centos 7

nahimov, 2020-06-22:
Hello! Could you please tell me i try to install on Centos 7 but i have problem with mp3lame, i just try to use different versions but have no luck, when i try make RTLSRD-Airband i have error:

`[root@localhost RTLSDR-Airband-3.1.0]#  make PLATFORM=x86
g++ -rdynamic  rtl_airband.o input-common.o input-helpers.o output.o config.o util.o mixer.o input-rtlsdr.o  -lrt -lm -ldl -lvorbisenc -lmp3lame -lshout -lpthread -lconfig++ -lfftw3f -lrtlsdr -o rtl_airband
/usr/bin/ld: cannot find -lmp3lame
collect2: error: ld returned 1 exit status
make: *** [rtl_airband] Error 1
`
I try to find at the system mp3lame and have no luck. All samples as i can find at internet only for Ubuntu

szpajder, 2020-06-22:
yum install lame-devel ?

nahimov, 2020-06-22:
Hi! Thank you! I have already installed lame-devel, but as i see this devel incorrect, i uninstall and reinstall again and all okay, thus trying configure.
Thank you!

szpajder, 2020-06-22:
I have built it successfully on CentOS 7.7 with the following dependencies fetched with yum:

lame-devel 3.100-1.el7
libshout-devel 1.4.9-5.el7
fftw-devel 3.3.3-8.el7
libconfig-devel 1.4.9-5.el7

nahimov, 2020-06-22:
All okay, i start rtlsrd-airband )) thank you! I have some problem with connecting airband with icecast, but i have victory! )) because some problem with language and understanding in manual. But right now all okay. I have a question. Rtlsdr-airband are goind through all frenq that i set to program or he watching like sdr# and Vasilis plugin at one bandwith 2MHz?

⁣Получить BlueMail для Android ​

п>I have built it successfully on CentOS 7.7 with the following

nahimov, 2020-06-22:
Yes i think that i answer about it, a have the problem with devel but i reinstall it and all okay

⁣Получить BlueMail для Android ​

п>yum install lame-devel ?

szpajder, 2020-06-22:
I have a question. Rtlsdr-airband are goind through all frenq that i set to
> program or he watching like sdr# and Vasilis plugin at one bandwith 2MHz?
>

It can do both - depending on how you configure it (search for multichannel
mode and scan mode in the wiki).

closed 2020-08-04 by szpajder