rtl-airband/RTLSDR-Airband #167: IceCast connection issue

AvrumFocus, 2020-07-05:
Hi,

I'm using your software and am loving it.

I set up a raspberry to stream to an icecast server I control and it's working fine. I set up another computer but somehow I keep on seeing the in the logs Could not connect to <snip>.

Any place where I can see more detailed logs?

szpajder, 2020-07-05:
If the raspberry and the new computer are both connecting to the same icecast server simultaneously, then you might have hit a limit on number of sources imposed by the Icecast server, which is set to 2 by default.

Inspect Icecast server logs (`/var/log/icecast/error.log` or similar). If the above is the case, increase `<sources>` parameter value in the `<limits>` section in the server config file and restart it.

AvrumFocus, 2020-07-05:
Thanks. I ruled out right away icecast sources limit on the icecast server.

I did a packet capture on the offending computer and see it is sending right away a tcp RST packet instead of an ACT packet. I don't know for sure it's rtl_airband sending that, but would like to rule that out.

Is there detailed logging on the icecast connection setup? or is there anyway I can enable debug logging?

szpajder, 2020-07-05:
What's the libshout library version on the client?

If it's 2.4.2 or later, then most probably you are hitting bug https://github.com/szpajder/RTLSDR-Airband/issues/143 . If this is the case, please downgrade to 2.4.1.

AvrumFocus, 2020-07-05:
Bingo. That's it. Thanks for the project and help.

closed 2020-07-05 by szpajder