rtl-airband/RTLSDR-Airband #171: Audio output highpass/lowpass help

poblabs, 2020-07-27:
When using audio output to mp3 file, I've tried every combination I can think of for highpass and lowpass to try and eliminate the "buzz" within the attached file. Nothing seems to work. 

If I move this mp3 into Audacity and do a highpass 1000, it works. From RTLSDR-Airband, a highpass or lowpass of 1000 does nothing. 

I'm using a high ampfactor because without it, the mp3 is extremely quiet.

Thoughts?

Here is a sample mp3 output: [highpass-help.zip](https://github.com/szpajder/RTLSDR-Airband/files/4983714/highpass-help.zip)

```
mixers: {
  mixer1: {
    outputs: (
      {
        type = "file";
        directory = "/home/pi";
        filename_template = "TESTOUTPUT";
      }
    );
  }
};

devices:
({
  type = "rtlsdr";
  index = 0;
  gain = 20;
  centerfreq = 156.5;
  correction = 40;
  channels:
  (
    {
      freq = 155;
      modulation = "nfm";
      lowpass = 1100;
      outputs: (
        {
          type = "mixer";
          name = "mixer1";
          ampfactor = 6.0; # 600% volume
        }
      );
    }
  );
 }
);
```

charlie-foxtrot, 2020-07-27:
Looks like a CTCSS tone at ~180Hz - https://en.wikipedia.org/wiki/Continuous_Tone-Coded_Squelch_System

You should be able to find the exact value from https://www.radioreference.com in the Tone column (ie 179.9 PL) 

Try using the unstable branch that includes https://github.com/szpajder/RTLSDR-Airband/pull/159

You shouldn't need to set `notch_q` (and `notch_gain` has been removed, so no need to set that either).

I would recommend setting both `highpass` and `lowpass` to `0` (if you don't set them they will default to `100` and `2500` respectively)

poblabs, 2020-07-27:
Thanks! I'll give the unstable branch a try. Checking radioreference, I see `179.9 PL` in the Tone column. Should I set `notch` to `179.9`?

charlie-foxtrot, 2020-07-27:
yes.  And I was wrong about `highpass` and `lowpass`, set them to `-1` 

```
    {
      freq = 155;
      modulation = "nfm";
      lowpass = -1;
      highpass = -1;
      notch = 179.9;
      outputs: (
      ...
```

poblabs, 2020-07-27:
I think that did the trick! I'll keep listening to the outputs but so far so good. Thanks!

closed 2020-07-27 by poblabs

poblabs, 2020-08-04:
@charlie-foxtrot Is it possible to apply the notch while in scan mode? I am scanning 2 frequencies and they both have different notches.

charlie-foxtrot, 2020-08-04:
> @charlie-foxtrot Is it possible to apply the notch while in scan mode? I am scanning 2 frequencies and they both have different notches.

it's possible but not currently supported.  open a new issue to add the support

poblabs, 2020-08-07:
Done thanks!