rtl-airband/RTLSDR-Airband #172: Possible to filter CW (morse code)?

poblabs, 2020-07-29:
My HT filters the CW (morse code), is it possible to have RTLSDR Airband do the same? I'm monitoring a low-usage frequency, but the CW is every hour causing a short recording every hour. I'd like to keep it to voice traffic only.

szpajder, 2020-08-04:
There is no such a feature in the program.

You can of course use notch filter to silence the morse code, but the squelch would still open, so a recording would be made anyway.

Is this such  a big deal? Storage is cheap these days.

poblabs, 2020-08-04:
> Is this such a big deal?

Nope, just annoying is all; I guess silence is better though. I can give the notch a try anyways to try and get rid of it. Any hints on what the notch value would be?

szpajder, 2020-08-04:
There is no universal CW frequency. Find it by trial and error or open a recording in Audacity (or any other audio editor), make an FFT and find out where its peak is.

poblabs, 2020-08-04:
Will give it a shot, thanks

closed 2020-08-04 by poblabs