rtl-airband/RTLSDR-Airband #173: Configuration for gain and prevent strong signal

pierrehenrymuller, 2020-08-06:
Hello,
I use rtl_airband with an hackrf.
I have found a good configuration for gain to listen most trafic exchange between tower and planes on local, medium range. But when a plane is too close the signal is too strong and incomprehensible.

I have test for gain = "auto" but I'm loosing planes who are far away.

How can I solve this by having a high sensitivity to pick up distant aircraft and having a mechanism that reduces the gain when the signal is too strong?

I probably missed a parameter or a concept in the wiki, which is why this question.
Thank you in advance

szpajder, 2020-08-06:
You can't solve this. The receiver has a finite dynamic range and your signals exceed that.

It's basically like taking a photo of a high-contrast scene. Either you get highlights exposed correctly, but then shadows are completely black or you expose for the shadows, but then your highlights are blown out. If the received signal is clipped, the software can't do much to "unclip" it.

pierrehenrymuller, 2020-08-10:
Ok I understand, there is a way like a module between antenna and hackrf I can plug to reduce the range of acceptable power?

And where I can find the value of this dynamic of sensibility for a product? What is the name of this indication for comparaison between receivers?

szpajder, 2020-08-10:
You can turn down the receiver gain or use an attenuator to reduce the chance of overloading the receiver, however this will attenuate all transmissions (including weak ones) by the same factor.

What you want is high dynamic range, which is a function of the number of bits of the analog-to-digital converter in the receiver. HackRF has an 8-bit ADC, which gives a theoretical dynamic range of approx. 50 dB. For comparison, a 12-bit ADC would have about 74 dB of dynamic range, while a 14-bit one would have 86 dB.

pierrehenrymuller, 2020-08-18:
Thanks for explainations, it's clear.
Do you know which SDR receiver models have 12 or 14 bits ADC?

pierrehenrymuller, 2020-08-18:
Ok I found some sdr receiver like RSPduo or RSPdx. I will see if there is supported by Soapy for integration with rtl_airband.
Thanks

rvt, 2020-10-09:
Could you also use two cheap RTL-SDR. The one for tower with a small antenna, and one for planes further away with a bigger antenna? Then in software switch between the two audio signals.

pierrehenrymuller, 2020-10-09:
Hum I understand this but have rtl_airband an option to switch between signal in diversity mode?
Or which software can make a diversity from two sdr?

rvt, 2020-10-09:
I might be facing the same issue as we are going to install something similar on location. I already route the audio through pulseAudio and icecast so I can add additional filters. When we have the same issue and use two SDR's I might work on a solution for this but by switching pulseAudio channels. Will update if I have something