rtl-airband/RTLSDR-Airband #175: RTLSDR-Airband with SoapySDR and AIRSPY HF+

theob24, 2020-08-06:
Hello,

I'm trying to get RTLSDR-Airband running with SoapySDR and AIRSPY HF+ on debian.

Everything is installed and the device is detected:

root@debian:/home/theo# SoapySDRUtil --probe="driver=airspyhf"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=airspyhf

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=AirspyHF
  hardware=AirspyHF
  serial=c852948094393f93

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CF32, CS16, CS8, CU16, CU8
  Native format: CF32 [full-scale=1]
  Antennas: RX
  Full gain range: [0, 54] dB
    LNA gain range: [0, 6, 6] dB
    RF gain range: [-48, 0, 6] dB
  Full freq range: [0.009, 31], [60, 260] MHz
    RF freq range: [0.009, 31], [60, 260] MHz
  Sample rates: 0.912, 0.768, 0.456, 0.384, 0.256, 0.192 MSps

I've this configuration file:

devices:
({
  type = "soapysdr";
  device_string = "driver=airspyhf,device_id=0";
 # gain = 25;
  centerfreq = 123.125;
 #correction = 0;
 modulation = "am";
  afc = 1;
  channels:
  (
    {
      freq = 123.125;
      outputs: (
        {
          type = "icecast";
          server = "***************";
          port = 8000;
          mountpoint = "TWR";
          name = "Tower";
          genre = "ATC";
          username = "source";
          password = "********";
        }
      );
    }
  );
 }
);

And when i'm running rtlsdr-airband in foreground, i'm seeing this:

 123.125

 108/ 90
 110/ 90
 110/ 90
 109/ 90
 110/ 90
 110/ 90
 110/ 90
 109/ 90
 108/ 90
 111/ 90
 110/ 90
 110/ 90

But when listening to the icecast stream, I don't have any sound.  It's not an Icecast issue because i'm able to stream sound with 
 an Icecast client, and using the same device (+ antenna) is working in sdr#. Did I missed something in the configuration file?

Thanks

charlie-foxtrot, 2020-08-06:
- try outputting to a file as well to rule out icecast
- when there is a signal detected there will be an asterisk to the right of the text column (ie `109/ 90 *`, if you dont see an asterisk, it isnt detecting signal
- `110 / 90` seems way to high, try dropping the gain (im working with a "bottom" number in the 10-20 range), check older issues for a description of what the numbers mean

starvald, 2020-08-09:
i think there is a glitch with airspy devices somewhere, with my R2 when i start it i can see high values and need to restart to get it to normal.

High Values:
162.250   161.225                                                             

  20/ 19    19/ 17  
  20/ 19    18/ 17  
  20/ 19    20/ 17  
  20/ 19    18/ 17  
  21/ 19    18/ 17  
  20/ 19    17/ 17  
  21/ 19    18/ 17  
  21/ 19    19/ 17  
  20/ 19    19/ 17  
  22/ 19    18/ 17  
  20/ 19    19/ 17  

What i believe is Normal:
 162.250   161.225                                                             

   1/  4     1/  3  
   1/  1     1/  1  
   1/  1     1/  1  
   1/  1     1/  1  
   1/  1     1/  1  
   1/  1     1/  1  
   1/  1     1/  1  
   1/  1     1/  1

theob24, 2020-09-13:
Everything is working fine with this configuration file:

devices:
({
  type = "soapysdr";
  device_string = "driver=airspyhf,serial=your_serial";
# channel = 0;
#  gain = 15.0;
# gain = "LNA=15,MIX=15,VGA=5";
# correction = 0.0;
# antenna = "Antenna A";
# mode = "multichannel";
# sample_rate = 2.56;
  centerfreq = 118.100;
  channels:
  (
    {
      freq = 118.100;
squelch = 30;
      outputs: (
        {
          type = "icecast";
          server = "xx.xx.xx.xx";
          port = 8000;
          mountpoint = "TWR";
          name = "Tower";
          genre = "ATC";
          username = "source";
          password = "xxxxxxx";
        }
      );
    }
  );

});

closed 2020-09-13 by theob24