rtl-airband/RTLSDR-Airband #177: [Feature Request] Raw output to file

lilrags16, 2020-08-11:
[Feature Request] Rawoutput to file

Concept:

- Each channel could be configured to use a different "sink"

- Raw output is dumped to a file so that external applications can interface with the data

This would be extremely useful to negate the need to use pulse/alsa to create virtual sinks

szpajder, 2020-08-12:
You are probably looking for https://github.com/szpajder/RTLSDR-Airband/wiki/Configuring-rawfile-outputs .

lilrags16, 2020-08-13:
Unfortunately that will not work as I would have to rewrite must the functionality of RTL-Airband

Brandon Ragsdale

szpajder, 2021-10-18:
Version 4.0.0 supports uncompressed audio streams over UDP, so maybe this suits your needs?

https://github.com/szpajder/RTLSDR-Airband/wiki/Configuring-UDP-outputs

ge0metrix, 2022-01-24:
UDP is great if you want to spin up other processes to capture it in real time. I would love to capture raw demodulate audio. 

The RawFile option is great, but requires demodulation after capture, we've already got a demodulator running, its kind of silly to have to demodulate again after the fact. 

1000ft overview of my thinking, at some point the raw demodulated audio is passed to LAME to encode and compress as an MP3, instead it could be handed to a file handler and be done?