rtl-airband/RTLSDR-Airband #185: Could not connect to BROADCASTIFYSERVER:80/MOUNT

dougware, 2020-09-07:
Hello,

I'm troubleshooting a problem with rtl_airband on Ubuntu 20.04 on a Raspberry Pi 3B.  I compiled rtl_airband using these flags, because Ubuntu 20.04 doesn't include some of the bcm_host headers/libraries.
`NFM=1 PLATFORM=armv7-generic`

I have verified the icecast details in the config several times.

All I'm seeing in the output is:
`Trying to reconnect to BROADCASTIFYSERVER:80/MOUNT...`
`Could not connect to BROADCASTIFYSERVER:80/MOUNT...`

I turned debugging on, but I didn't see any connection related information at all in the debug_log.  I can reach the Broadcastify server through curl, but I get a 404.  Hitting an icecast server mountpoint with curl may normally return a 404, I'm not sure. 

How can I see the icecast connection failure details?

dougware, 2020-09-07:
I went back to the latest Raspbian for this Pi and rtl_airband is working as expected.

I didn't compile any libraries on the Ubuntu Server 20.04 Pi image, I used stock packages for everything until I compiled rtl_airband.  I'm not sure why there was a difference.

Doug

szpajder, 2020-09-07:
My guess is that you are hitting issue https://github.com/szpajder/RTLSDR-Airband/issues/143.

closed 2020-09-08 by dougware