rtl-airband/RTLSDR-Airband #188: Mixer enhancement

ThellraAK, 2020-09-14:
I'm trying to set up a stream for something like broadcastify, where I'd stream Police, Fire, EMS, etc

It would be helpful to be able to have the mixer have priorities, so if the two (or more) channels are active at the same time, one wouldn't be played, with the any number of mixers option, this would be really nice, as it would be possible to have different streams with different priorities.

My attempt at an example

devices: ({
  ...
  channels: ({
    freq = 120.875;
    outputs: ({
      {
        type = "mixer";
        name = "mixer1";
\#       balance = 0.0;
\#       ampfactor = 1.0;
\#     priority = 1
      }
    });
  });
});

then in mixer1 if there were multiple outputs active at the same time, all (or another option for how many are allowed at once?) but the highest priority is suppressed.

thowe-switzerland, 2021-07-12:
I have a similar need: A kind of scanner but not on the radio frequency but on several demodulated audio inputs of a mixer. So that not all channels are mixed together, but only one is output with activity picked out.

I have made an experimental implementation on a separate branch. Everything is still quick & dirty. But it works already.

Maybe I will find some time during the vacations and try to provide a clean branch of such a "scanning mixer".

Translated with www.DeepL.com/Translator (free version)

rpatel3001, 2022-04-30:
+1 with an additional request

I want to mix a bunch of channels into one output and have the highest SNR channel take priority

charlie-foxtrot, 2022-04-30:
Sounds like the ask isn't to mix anything, but rather have a "selector" that is similar to a "mixer" except selects a single active input to pass to the output based on something (priority, SNR, etc).

May be odd to have one active channel interrupted mid-transmission by another channel of higher priority coming active.